mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #255: ESP32-S2 compile errors, missing #include "esp32-s2_i2s_parallel_v1.h"

wibenjamin, 2022-02-12:
Hello,
I try to compile V2.0.6 for ESP32-S2 in Arduino IDE V1.8.19 and getting some errors like this:

```
esp32_i2s_parallel_v2.h:51:18: error: 'I2S_NUM_1' was not declared in this scope
       if(port == I2S_NUM_1) {

```

I could figure out ESP32-S2 only has one I2S port so this error makes sense to me, I also saw the include of the headerfile "esp32-s2_i2s_parallel_v1.h" in "ESP32-HUB75-MatrixPanel-I2S-DMA.h" but it seems to be missing in the source files. Can you add it to the repository? Is there a problem somewhere else?

Thanks and best regards.

solhuebner, 2022-02-12:
@wibenjamin can you actually try to use the current master?
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/archive/refs/heads/master.zip
There were quite a few changes from that since 2.0.6 was released :)

wibenjamin, 2022-02-12:
Thanks, now is compiling fine!

closed 2022-02-12 by wibenjamin