mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #261: Has anyone fixed the life pattern ?

tablatronix, 2022-03-06:
Was just curious, cause it is clearly wrong. Maybe bad neighbor code, bad random entropy?

It always winds up with these repeating lines and lots of symmetry

[image]

mrfaptastic, 2022-03-09:
Not sure what's going on with these aurora examples, the code isn't maintained and may need to be refactored.

tablatronix, 2022-03-09:
Ill work on them , did these come from fastled? I noticed alot have been modded time use effects.h

I have some working and some working better

mrfaptastic, 2022-03-12:
The code was originally coped from here: https://github.com/pixelmatix/aurora

closed 2022-11-09 by mrfaptastic