mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #271: Pproblem with GFX fonts

zznty, 2022-03-26:
+ I have read and tried suggestions from #134 but nothing changed
+ Issue persists in 3 different esp32 boards
+ Drawing pixels and geometry is wroking fine
+ Tried to use other fonts from gfx library but all of them having same issue

[image]

umutinevi, 2022-06-15:
Same here. Fonts doesn't seem to work properly.

1/8 scan rate 64x32 panel here.

All fonts give a result of blank pixels or mislocated pixels.

solhuebner, 2022-07-04:
@zznty could you solve the issue?

Can you otherwise post an example code so that we can try to replicate it?

zznty, 2022-07-06:
nope.
Havent tried new updates, last attempts was in march.
the main problem if im using the gfx api such as `print` or `println`, the font getting damaged
but it's fine with displaying gifs, plasma and other shapes

10der, 2025-08-05:
hi!

sorry for hijacking.

MRA:
```

#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

#define PANEL_WIDTH 128  // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_HEIGHT 64  // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1    // Total number of panels chained one to another

//MatrixPanel_I2S_DMA dma_display;
MatrixPanel_I2S_DMA *dma_display = nullptr;

void setup() {
  HUB75_I2S_CFG mxconfig(
    PANEL_WIDTH,   // width
    PANEL_HEIGHT,  // height
    CHAIN_LENGTH   // chain length
  );

  mxconfig.gpio.e = 32;
  mxconfig.clkphase = false;
  mxconfig.driver = HUB75_I2S_CFG::ICN2038S;
  mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_20M;

  // Display Setup
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(90);
  // dma_display->clearScreen();
  //dma_display->setFont(&Picopixel2);
  dma_display->setTextSize(1);
  dma_display->setTextColor(dma_display->color444(255, 255, 255));

dma_display->setTextWrap(false);

dma_display->setCursor(1, 1);
dma_display->print("ABCDEFGHIJKLMNOPQRSTUVWXYZ");
dma_display->setCursor(1, 8);
dma_display->print("1234567890");
dma_display->setCursor(1, 16);
dma_display->print("ABCDEFGHIJKLMNOPQRSTUVWXYZ");
dma_display->setCursor(1, 24);
dma_display->print("1234567890");
dma_display->setCursor(1, 32);
dma_display->print("ABCDEFGHIJKLMNOPQRSTUVWXYZ");
dma_display->setCursor(1, 40);
dma_display->print("1234567890");
dma_display->setCursor(1, 48);
dma_display->print("ABCDEFGHIJKLMNOPQRSTUVWXYZ");

}

void loop() {
}
```

the same as author topic starter.

[image]

board707, 2025-08-05:
I think that the Adafruit library and fonts have nothing to do with it. Considering that the first text line in the picture is shown correctly, there are no problems with extracting letters from the font.

What we see on the screen looks more like a swapping in the order of outputting pixel rows, probably caused by inaccurate operation of the multiplexer, mismatch of logical signal levels or phase of control pulses.

10der, 2025-08-05:
> I think that the Adafruit library and fonts have nothing to do with it. Considering that the first text line in the picture is shown correctly, there are no problems with extracting letters from the font and printing them on the panel.
> 
> What we see on the screen looks more like a swapping in the order of outputting pixel rows, probably caused by inaccurate operation of the multiplexer, mismatch of logical signal levels or phase of control pulses.

thank you. any recommendation what i should check?

10der, 2025-08-05:
> mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_20M;
no. Nothing change. alas.

>Do you have another panels, with different model, another chips on it? Is the problem reproducible on other matrices?
no. :(

it's from "ali"
P2.5 LED Display RGB Module 128x64,1/32 Scan
SMD1515
RUC7258
driver ic is ICN2037, decode 138
driver ic is ICN2037, decode 1e8