mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #272: split data

harissutanrafiq, 2022-03-26:
how i can identified or split a data is for upper (rows) or for lower (rows) if i use 128 x 64 pixel..?

board707, 2022-04-06:
Please explain, why do you need this?

harissutanrafiq, 2022-04-08:
this just for POC after i see hiudu controller or enother controller in the market using more than one HUB75

DarrylStrong, 2022-04-09:
I did similar to use 16s panels with 64 pixel height displays but used custom hardware to switch the clock line between output ports.

As the D line isn't used by the 16s panels that performs the switch function.

harissutanrafiq, 2022-04-09:
> I did similar to use 16s panels with 64 pixel height displays but used custom hardware to switch the clock line between output ports.
> 
> As the D line isn't used by the 16s panels that performs the switch function.

i just add diferent lat pin, enother pin is still same

harissutanrafiq, 2022-04-09:
this the code i have modified
-------------------------------------------------------
ESP32-HUB75-MatrixPanel-I2S-DMA.h
 [https://pastebin.pl/view/943cff26
](url)  

ESP32-HUB75-MatrixPanel-I2S-DMA.cpp
[https://pastebin.pl/view/99450647](url)

board707, 2022-04-09:
Why outputs the top and bottom rows on different HUBs ?  The existing software supports the connection of several panels in a chain, including two panels on top of each other. If you switch clock or lat - you still get the output sequential and takes exactly the same time as the output in one chain, that is, your method does not give any advantages - it's just a complication.

harissutanrafiq, 2022-04-09:
> Why outputs the top and bottom rows on different HUBs ? The existing software supports the connection of several panels in a chain, including two panels on top of each other. If you switch clock or lat - you still get the output sequential and it takes exactly the same time and the same memory as the output in one chain, that is, your method does not give any advantages - it's just a complication.

first its just POC i just want to know how its work, and why huidu and other running controller in the market split the hub if that not have any andvantages?

board707, 2022-04-09:
> why huidu and other running controller in the market split the hub if that not have any andvantages?

Split the matrix to several HUBs has their advantages if you outputs signal to several HUBs in parallel. But for that you need to use separate sets for all data pins (R0 R1 G0 G1 B0 B1) for each HUB and, of course, compatible software.

DarrylStrong, 2022-04-09:
With another latch pin the timing gets odd with the rows changing. I used that method first but it meant lower brightness.

The reason I had to go to two ports is that the chain would have been 22 modules long which was causing glitching and also that I have to be able to switch between my cpu driving the modules and a standard novastar receiver card.

harissutanrafiq, 2022-04-13:
hi @board707 @DarrylStrong thanks for your respon.