mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #273: Function to transfer the whole pannel(s)?

zesinger, 2022-03-30:
Hello
Is there any fast function to transfer the whole matrix pannel(s) in one call? I have a full buffer filled with RBG values and would like to avoid the `for (j=...){for (i=...){dma_display->drawPixelRGB888(ti,tj,R,B,G);}}` if possible.
Thanks
David

vortigont, 2022-04-06:
No, there is no such methods for now except fast functions to draw lines and rectangles.
I was thinking about creating exactly the thing you are asking for, but it's not that easy as it may seem. For one panel is not a problem, but for chained and combined panels it requires sophisticated transformation logic that ruins the whole idea.
Still this idea is my TODO list for future :) (do not have those RGB panels in possession for now)

zesinger, 2022-04-06:
Thanks a lot for your answer.
Even x calls for x pannels could be great!

board707, 2022-04-06:
>  it's not that easy as it may seem. For one panel is not a problem, but for chained and combined panels it requires sophisticated transformation

It is not difficult for directly chained panels and even for zigzag connections, no transformation required. I tryed to program fast functions for running text by shifting entire pixture to left or right.

zesinger, 2022-04-07:
Thanks anyway for your great software.

closed 2022-05-15 by mrfaptastic