mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #274: limitation daisy chain

emhatek, 2022-04-06:
Dear mrfaptastic,
based on your experience for creating this library, how many panel can be daisy chained at the maximum. (column and row).

thanks

DarrylStrong, 2022-04-06:
Hi,

I have 384x64 pixels running on our signs and have tested to 512x64.

This is using reduced bit depth.

Cheers

Darryl

ergindemir, 2022-04-06:
> 

Have you ever tried a wroom module with psram? Would it make a difference?

ergindemir, 2022-04-06:
[DarrylStrong](https://github.com/DarrylStrong) Have you tried a wroom module with psram, would it increase the ram available to the DMA?

DarrylStrong, 2022-04-06:
I use an ESP32-WROOM-32D, mounted onto a custom PCB to provide level shifting and also logic switching to allow me to use 16s modules pretending to be 32s for the library.

zesinger, 2022-04-07:
Hello
Do you think that using 4 64x64 pannels -> 256x64 would work in RGB24?

vortigont, 2022-04-07:
> Do you think that using 4 64x64 pannels -> 256x64 would work in RGB24?

@zesinger 
it will, I've tested it some time ago. But refresh rate will be less than acceptable 100 Hz.
You can use this calc to estimate mem/color/refresh rate options 
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/master/doc/i2s_memcalc.md

zesinger, 2022-04-11:
@vortigont 
Thanks for your answer. I really don't need 100Hz as I have coded a real dot matrix display for virtual pinball emulators. The "video" displayed on it are something like 20Hz max. [https://www.pincabpassion.net/t14796-tuto-zepindmd-installation-english](https://www.pincabpassion.net/t14796-tuto-zepindmd-installation-english)

vortigont, 2022-04-11:
@zesinger this is awesome project! :))) Nostalgic :) love it!
100Hz is not a frame rate it's a matrix refresh rate. Even static picture will flicker like a strobo at 20Hz. By default lib does not allow you to go below 100 Hz, it will reduce color transition automatically. To reduce the rate you need to build it with a redefined min_rate. But I suggest to set your color depth based on estimation with a spreadsheet I've referenced above. You will also benefit with a reduced DMA buffer size in that case. True 24 bit is almost impossible for anything above 64x64 at 1/32 scan due to the physical limits of the shift registers, it works at 20-25 MHz max.

zesinger, 2022-04-12:
Do you mean that I should send pixels with something else than "drawPixelRGB888" to make 4x64x64 panels to work without flickering? Sorry, I didn't find any list of all the functions available in your library

vortigont, 2022-04-14:
@zesinger you can find all the methods in [header](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/master/ESP32-HUB75-MatrixPanel-I2S-DMA.h) file.
You should use calc from [here](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/master/doc/i2s_memcalc.md) to find your optimal bit depth, than build the lib with `PIXEL_COLOR_DEPTH_BITS` defined to that value. Than use any method you like to draw pixels, `drawPixelRGB888` or `drawPixel` will result the same.

zesinger, 2022-04-14:
@vortigont OK I'll do so, thanks for your time!