mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #277: ESP32 PICO implimetation

mithy99, 2022-04-22:
I am very beginner of ESP32 and I am trying to impliment this library into ESP32 pico mini2 board

compile is fine but after I download the code into the board, it keep rebooting

I think the problem is comes from 
the 83 line of esp32_i2s_parallel_dma.c
   --> PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[gpio], PIN_FUNC_GPIO);

since the pin configuration of ESP32 chip on WROOVER and PICO is different,

is there any way can I fix?

mithy99, 2022-04-22:
serial log is like below

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x2aaa0008,len:0
ho 12 tail 0 room 4
load:0x00000000,len:-2147483648
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e4d27  PS      : 0x00060933  A0      : 0x800e47d4  A1      : 0x3ffb1dc0  
A2      : 0x00000017  A3      : 0x00000003  A4      : 0x00060520  A5      : 0x00000020  
A6      : 0x00000020  A7      : 0x3ffc2b38  A8      : 0x6abaaaaa  A9      : 0x00013ffc  
A10     : 0x00000001  A11     : 0xaaaaaaaa  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060920  A15     : 0x00000000  SAR     : 0x00000009  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xaaaaaaaa  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400e4d27:0x3ffb1dc0 0x400e47d1:0x3ffb1df0 0x400d137d:0x3ffb1e20 0x400d151a:0x3ffb1e40 0x400d1c6d:0x3ffb1e80 0x400d113b:0x3ffb1f50 0x400d3776:0x3ffb1fb0 0x40086115:0x3ffb1fd0

Rebooting...

zesinger, 2022-04-22:
Could you show your code?
Are you sure you initialize well before trying to draw something?
Did you try the examples? Fist try to make the example https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/master/examples/1_SimpleTestShapes to work, then keep the panel configuration and code from that.

ergindemir, 2022-04-22:
Which devkit are you using? Do you use the standard pins defined in the header file?

mithy99, 2022-04-22:
Thank you for reply
And I am trying the sample code and still not working..

Same problem

mithy99, 2022-04-22:
And I am using esp32 pico mino v2  board

mithy99, 2022-04-24:
I checked the datasheet of esp32 pico chip and I think this chip doesnt support I2S parallel mode.

closed 2022-04-24 by mithy99