mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #283: Question about Panels 64x64 Pixel, 1/32 Scan LED Matrix.

burkadius, 2022-05-20:
I use this panel in this project https://www.instructables.com/Talking-Tetris-Clock/
Unfortunately, the panel does not work at all. The library HUB75 RGB LED matrix library utilizing ESP32 DMA supports the following panel driver chips and 64x64 matrix,1/32 scan.
ICND2012
RUC7258
FM6126A AKA ICN2038S, FM6124 (Refer to PatternPlasma example on how to use.)
SM5266P
On my panel (Aliexpress) P2.5 64x64-32S-V2.0 the following chips are installed
CHIPONE ICND 2153AP
MW245B
FM - TC7258GN
74HC04S
Is there any experience with this chipset? Should it work?
Kind regards
Burkhard

board707, 2022-05-23:
ICND2153 is a driver with s-PWM technology, such drivers are not supported by the library.

board707, 2022-05-31:
The datasheet of the ICND2153 is quite brief, but the MBI5153 and 5152 drivers are very similar to the one you are looking for and their datasheets can be a useful source of information.
[https://datasheetspdf.com/pdf-file/974150/Macroblock/MBI5153/1](MBI5153)
[https://www.neumueller.com/download/MBI5152%20Anwendungshinweis%20-%20Application%20Note.pdf](MBI5152 App_Note)

board707, 2022-06-01:
For most china LED panels the only available documentation are datasheets of the drivers.

ergindemir, 2022-06-01:
If there is no doc how does the control card manufacturer designs the board? It must be somewhere.

board707, 2022-06-01:
maybe but I haven't seen it

ergindemir, 2022-06-19:
I have tested the 64x64 p3 panel with 4-4.2V power supply, the issue still persists. I think upon this observation we can conclude that it's not a logic buffer issue.

dosipod, 2022-06-27:
We got 4 panels that are P2 64X64  1/32 Scan ,We do not also have an accurate datasheets but some of the examples worked with few modification for 64x64 . Can you test  the examples in the link below . If they worked for you even partially then we have something to go on for further work  .we just got those and for sure we can  try to chain as much as possible and test for others as we have a lot  of MCUs and HW  and this is part of students projects and  we have no real use case other then learning 

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/issues/288#issuecomment-1166697540

pryankster, 2023-05-26:
I found [this archive](https://www.dropbox.com/s/84ccorlzdu07395/LED%20Datasheets%20CN.zip?dl=0)  on a chinese document website. it contains application notes, registers, and timing information for a bunch of different driver chips.  The docs are all in chinese, but should be translatable with google translate.

The text along with it translates to:  "Common LED control driver chips with register setting information: such as icn2038s, icn2053, mbi5151, mbi5152, mbi5153, mbi5169, mbi5170, mbi5224, mbi5324, mbi6023, etc."

I'm poring over it right now, hopefully the data will be useful for others, too.

mrfaptastic, 2023-05-31:
If you turn any of your investigation into code this library could use it will be most appreciated.