mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #285: How to increase the brightness

seoungbin0125, 2022-05-25:
Is there a way to increase the brightness?
It is dark even though it is raised to the maximum through the setPanelBrightness() function.
The left side of the picture below is the corresponding library, and the right side is the other library.
Please give me some advice.

[image]

DarrylStrong, 2022-05-25:
What colour bit depth are the two libraries set to?

seoungbin0125, 2022-05-25:
The color bit uses the color565 to 888 function.
I don't know what the hardware says
How can I set the color bit?
The other library is p3rgb

DarrylStrong, 2022-05-26:
I was meaning the libary setting for output colour resolution. 

// 8bit per RGB color = 24 bit/per pixel,
// might be reduced to save DMA RAM
#ifndef PIXEL_COLOR_DEPTH_BITS
 #define PIXEL_COLOR_DEPTH_BITS      **8**
#endif

closed 2022-05-30 by seoungbin0125