mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #286: Single panel issue with 1/8 scanning

alba-ado, 2022-06-07:
Hello,

I've been using a 64x32 LED Matrix display with 1/8 scanning. I couldn't seem to get your library to work with the 1/8 example. However, as I was trying to make it work I found out that the "ESP32-VirtualMatrixPanel-I2S-DMA.h" has some code in it;

`// Stupidity check
	if ( (vmodule_rows == 1) && (vmodule_cols == 1)) // single panel...
	{
		coords.x = x;
		coords.y = y;			
		return coords;
	}`

which checks if there is a single panel and returns normal coordinates, but since it doesn't do any change on the coordinates, it doesn't work with a 1/8 display. However, when I removed this part of the code, it worked. I wanted to create a pull request with this part removed, but I don't know what effect it will have on other users. For me, it seems to work with 1 and 2 panels with this part removed. Otherwise it doesn't work. Can you check if we can remove this part or can you make any other suggestion to fix this issue?

Thanks for your help.

board707, 2022-06-07:
For 2 panels you do not need to remove this part of code, it should work anyway (because of the fact that "stupidity check" condition is not true).
In case of single panel would be better do not remove the code but slightly edit the condition:

if ( (vmodule_rows == 1) && (vmodule_cols == 1) && ( _panelScanRate != ONE_EIGHT_32)) // single panel...

alba-ado, 2022-06-07:
What does this check do? I can't seem to find any functional difference when I remove it. And why you don't want to run the rest of the code for a single panel?

board707, 2022-06-07:
alba-ado, I'm not the author

solhuebner, 2022-06-08:
I can push a PR over the weekend. The addition of the special panel types broke the flow as it seems.

Red-77, 2022-06-29:
> What does this check do? I can't seem to find any functional difference when I remove it. And why you don't want to run the rest of the code for a single panel?

since the 1/8 scan example was designed for chained panles, what these lines do is that if only 1 display is defined, it just ignores it and dosent modify the coordinates to make it compatible with 1/8 panles. honestly i myself dont understand why it was done but what you can do is add 2 copies of the library one that is modified and one that is stock.

alba-ado, 2022-06-29:
I think you mean compatible with 1/16 panels? If I understand it correctly, you can set it to be 1/16 displays by giving the display type at the beginning anyways, so I don't understand what this code do. Can we remove it if it is not affecting anyone?

alba-ado, 2022-06-29:
I don't see it doing anything useful. Can you explain to me why it is there? I don't see any use scenario where this would be useful.

alba-ado, 2022-06-29:
It doesn't change anything about how the displays are driven. It only looks for the number of displays, and doesn't touch the pixels if there is only 1 panel.

solhuebner, 2022-07-04:
Sorry for the delay but I just pushed PR #295 which should fix the logic flow issue.
Please let me know if you encounter any issues :)

PR #307 (2022-08-16): Fix VirtualMatrixPanel scan rates
New PR to include changes made by #305 

Fixes #286 and related single display issues

I tried to avoid "fixing" the formatting but it was all over the place so I had to ;)

Would be great if we could get some QC tests, especially for single display VirtualMatrixPanel usage.
commit 4cb2de09 (2022-08-16): Fix VirtualMatrixPanel scan rates

New PR to include changes made by #305 

Fixes #286 and related single display issues

I tried to avoid "fixing" the formatting but it was all over the place so I had to ;)

Would be great if we could get some QC tests, especially for single display VirtualMatrixPanel usage.
closed 2022-08-16 by mrfaptastic