mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #288: Do you support 128x64 panels?

seoungbin0125, 2022-06-14:
Do you support 128x64 panels?
Please let me know if there is anything I need to set up
Thank you.

It looks like this now.

[image]

board707, 2022-06-14:
Please provude the photo of the other side

dosipod, 2022-06-26:
hello , I have 4 panels each is p2 64x64 , so i was also hoping to make 128x64 or more .
I am not able to correctly workout how to use only one 64x64 . 
I started by modified ESP32-HUB75-MatrixPanel-I2S-DMA.h and added MATRIX_HEIGHT  64 and E_PIN_DEFAULT 32 but the result is not correct , i think i might have a wiring issue but also could be i did not modify the examples correctly . I will keep on trying until i get it because I only got the panels today :)

https://user-images.githubusercontent.com/87589371/175795661-cccb9985-dc80-47d6-a532-de40a5c50c69.mov

dosipod, 2022-06-26:
64x64 worked better with the other examples but still not perfect

https://user-images.githubusercontent.com/87589371/175799887-455811a3-94db-43ed-96a1-26a495052d43.mov

dosipod, 2022-06-27:
ok i was able to fix the examples for 64x64 , please see if you can also modify the same to fit yours until i get the extra cables to chain an extra 64x64 .

What I did is modify ESP32-HUB75-MatrixPanel-I2S-DMA.h  ( made it 64x64 )   . And then inside the examples I have copied the the function below from     2_PatternPlasma.ino  :
 
  HUB75_I2S_CFG mxconfig;
  mxconfig.mx_height = PANEL_HEIGHT;      // we have 64 pix heigh panels
  mxconfig.chain_length = PANELS_NUMBER;  // we have 2 panels chained
  mxconfig.gpio.e = PIN_E;    // we MUST assign pin e to some free pin on a board to drive 64 pix height panels with 1/32 scan

And added them to the other examples and they worked , here is a copy of all the modified  examples  that worked . Just a not that some of the examples will reboot the MCU but if you press gpio0 then the next effect will work ( I face crashes with "spiro " effect only )   . I hope that that will help .

[hub75.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/files/8987832/hub75.zip)

yangminglong, 2022-08-15:
My 96 * 128 panel works very well