mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #289: Help

zCrystal94, 2022-06-16:
Hi,
Can i use sdcard with this library? If yes, can you show me the pinout?
Thanks.

mrfaptastic, 2022-06-18:
I'm not sure there are enough GPIOs for both.

zCrystal94, 2022-06-20:
i am using this board: 
[image]
Can i use for sdcard with this pins?: 28,29,30,31,32,33

mrfaptastic, 2022-08-16:
This library needs 13 GPIOs (for 64x32 panel), and 14 GPIOs if it's a 64x64 pixel panel.

If I cross out the standard pins the library uses for a 64x32 pixel panel, you should have enough left over. You can't use the FLASH pins though.

[image]

mrfaptastic, 2022-08-16:
It looks like somebody has already done this, refer to the discussion:

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/discussions/155#discussioncomment-3009149

mrfaptastic, 2023-03-27:
There is now an animated gif SD card example with this libraries examples.