mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #292: How to change brightness with push buttons.

Red-77, 2022-06-28:
so first of all thank you for your amazing work, its the best hub75 library i have used.
So what my issue is that i cant change the panel brightness. Lets say i use the "PatternPlasma" example if i change the value of [dma_display->setBrightness8(192);] , nothing happens, so is there a way to actively change the panel brightness through out the code, i want to control the brightness via the serial monitor or push buttons to change the brightness

DarrylStrong, 2022-06-29:
Do you have OE connected properly? This is used to turn the panel off and therefore alter the brightness.

Red-77, 2022-06-29:
> Do you have OE connected properly? This is used to turn the panel off and therefore alter the brightness.

actually i figured it out, i just have to call the brightness function before i use any pattern, i just had put it in the wrong place, but im still confused about how to alter the brightness in real time, via serial monitor, slider, potentiometers or push buttons. can you help me out with that a bit

closed 2022-06-29 by Red-77