mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #293: class MatrixPanel_I2S_DMA' has no member named 'drawPixelRGB565'

Loufey, 2022-06-29:
src/main.cpp:106:24: error: 'class MatrixPanel_I2S_DMA' has no member named 'drawPixelRGB565';

how to fix this error

Red-77, 2022-06-29:
• its supposed to be like this i think

"dma_display->color565(r, g, b)"

• r,g,b are the values (0-255)

• for example

"dma_display->fillScreen(dma_display->color565(128, 0, 0));"

Loufey, 2022-06-30:
> • its supposed to be like this i think
> 
> dma_display->color565(r, g, b)
> 
> • r,g,b are the values (0-255)
> 
> • for example
> 
> dma_display->fillScreen(dma_display->color565(128, 0, 0));

Thanks.
I fond this function in the Souce Code ver2.0.0

inline void MatrixPanel_I2S_DMA::drawPixelRGB565(int16_t x, int16_t y, uint16_t color) 
{
  uint8_t r = ((((color >> 11) & 0x1F) * 527) + 23) >> 6;
  uint8_t g = ((((color >> 5) & 0x3F) * 259) + 33) >> 6;
  uint8_t b = (((color & 0x1F) * 527) + 23) >> 6;
  
  updateMatrixDMABuffer( x, y, r, g, b);
}

closed 2022-06-30 by Loufey