mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #302: Matrix Shield for ESP32 and horizontal lines not displaying

GrimorJander, 2022-07-20:
Hi everybody.
I just bought a RGB Matrix Shield for ESP32 dev module with two diferent wirings: Smartmatrix and P3RGB64X32 (Neocat)
[image]
Out of the Box is incompatible with this library, so i follow the wiring in order to find out the real pinout of both ports.

The result:

`//SMARTMATRIX
#define R1_PIN 2
#define G1_PIN 15
#define B1_PIN 4
#define R2_PIN 16
#define G2_PIN 27
#define B2_PIN 17
#define A_PIN 5
#define B_PIN 18
#define C_PIN 19
#define D_PIN 21
#define E_PIN -1 // required for 1/32 scan panels, like 64x64. Any available pin would do, i.e. IO32
#define LAT_PIN 26
#define OE_PIN 25
#define CLK_PIN 22
 
 //P3RGB64X32
//#define R1_PIN 25
//#define G1_PIN 26
//#define B1_PIN 27
//#define R2_PIN 21
//#define G2_PIN 22
//#define B2_PIN 23
//#define A_PIN 12
//#define B_PIN 16
//#define C_PIN 17
//#define D_PIN 18
//#define E_PIN -1 // required for 1/32 scan panels, like 64x64. Any available pin would do, i.e. IO32
//#define LAT_PIN 32
//#define OE_PIN 33
//#define CLK_PIN 15`

Well, then i modify the examples here in order to use custom GPIOs.
I´ve tried both ways, from the skecht with define i2S pins: `HUB75_I2S_CFG::i2s_pins _pins={R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN};`
and modifying the "ESP32-HUB75-MatrixPanel-I2S-DMA.h" file.

Both ways result in displaying with the same horizontal separation/missing  lines:
[image]
Any ideas of how to deal with this?
Thanks in advance

mrfaptastic, 2022-08-16:
Something is wrong with the B_PIN

secangkirkopipanas, 2022-10-21:
Hi @GrimorJander, any update on this issue? I'm facing kinda similar issue.
[](url)
[image]

Anyone can assist me on this issue? Thank you

GrimorJander, 2022-10-21:
@secangkirkopipanas 
I really didn't find a solution.
I have simply stopped using the board with this library.
The adapter works fine with other libraries, and when I modify the pinout in the code to make it compatible with this one, the problem appears.
Likewise, the library works fine with the default pinout, so I made my own adapter.
I only have 64x32 panels, so it´s more standar and i dont have to modify "ESP32-HUB75-MatrixPanel-I2S-DMA.h" file .

DarrylStrong, 2022-10-21:
Looks to me like the 'A' line is missing....

mrfaptastic, 2022-10-21:
> [](url)
> [image]
> 
That looks like the VirtualMatrixPanel test pattern? 

And it looks like everything is working perfectly, see the nice outer box?

Try not using the virtualmatrix class, you don't need it for just a horizontal chain.

mrfaptastic, 2022-10-21:
> Looks to me like the 'A' line is missing....

Agreed

And also looks like the use of the VirtualMatrixPanel class which is not required on a simple horizontal chain.

secangkirkopipanas, 2022-10-21:
Hi all, thank you for your response. My bad, I realized that I was using WROVER for this. Not an issue at all once I changed to WROOM.

secangkirkopipanas, 2022-10-21:
> 

This is just my simple testing on 64x64 chained panels. I will extend it to 6 panels with 2 rows and 3 columns.

GrimorJander, 2022-10-21:
> > My bad, I realized that I was using WROVER for this. Not an issue at all once I changed to WROOM.
> 
> Is the problem solved now?
> 
> By the way - why there are 4 HUB connectors on the shield - after all, neither library supports parallel output to 2 channels, as far as I know

You have to choose just one from 2 diferent pinouts (P3RGB64X32 or SMARTMATRIX) and two variants (3.3V or  5V with the level shifting buffers.
https://www.tindie.com/products/lightwell/rgb-matrix-panel-shield-for-esp32/
Seens good product (the only board available when i bought), and my only complain is can`t make it work with this library.