mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #303: is this library compatible with 1/32 scan?

lols, 2022-07-21:
Hi, is it compatible with 1/32 scan panel do i need to set scan or the library default scan is 1/16?

for 2 panel one top one bottom
```
#define PANEL_RES_X 128
#define PANEL_RES_X 128
#define NUM_ROWS 2 
#define NUM_COLS 1

#define SERPENT true
#define TOPDOWN false

virtualDisp = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, SERPENT, TOPDOWN);

```
is this correct?
Thanks

DarrylStrong, 2022-07-21:
You haven't defined PANEL_RES_Y.....

lols, 2022-07-21:
ouch hahaha copy paste wrongly

DarrylStrong, 2022-07-21:
I use the library to drive 384x64 1/32 displays. I don't use virtualmatrixpanel for this as it drives them natively. Due to the 64 pixel height it drives them at 1/32 automatically.

dosipod, 2022-07-31:
I have used the code with p2 64x64 1/32 scan , look at the code here  https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/issues/288#issuecomment-1166697540
@DarrylStrong  could you please share your code for 384x64 ? I have 4 other panels but only used one  ,

DarrylStrong, 2022-08-01:
I just set the panel res to 384x64 and one row/column.
Not easy to share my code as I have heavily modified the library from an older version.
Just make sure that you have all the output pins correctly allocated and the right driver chip selected.

mrfaptastic, 2022-08-16:
@dosipod - Can I close this issue?

closed 2022-09-12 by mrfaptastic