mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #308: P4 80x40 pixel 1/10 scan

tom3f, 2022-08-10:
Hello!
I've got some led's on the 1/10 outdoor panel (80x40 pixel) running, but the rows are not fired correctly. 
The panel has 60x ICN2037 and 8x ICN2012. 
Is there a trick to setup the adressing mode for 1/10 scanrate?

board707, 2022-08-11:
Hi!
Your matrix is a tricky case. Firstly, most libraries do not support 80x40, since it is quite rare. Although, in general terms, the operation of this matrices is similar to 64x64 or 128x64 - they also have 5 address lines A-E. 
Secondly, a standard 80x40 matrix has 1/20 scans, and you have 1/10. This means that the LEDs in your matrix are connected in a non-standard way.
If you have one such matrix, then you probably shouldn’t waste time and it’s easier to buy something more common.

tom3f, 2022-08-13:
Yeah, you are absolutly right - using those very cheap chineese outdoor panels that have zero support/documentation is a waste of time.

But, i did not give up and invested a whole day in this (reverse engineer schematics, write new driver, etc.) and got it working now with simple parallel bit-banging on a nodemcu-esp32.

And the led's are really bright...

[image]

ergindemir, 2022-08-13:
Good job, is it 160x20 internally? Did you modify the library?

board707, 2022-08-13:
> is it 160x20 internally? 

Very interesting question.  I hope @tom3f tells us

@ergindemir 
 160x20 - Do you mean 80px of one row first and then 80px of the other?
But I don't think so. Outdoor panels almost never use this type of connection. In this case, there would be too long tracks on the board. Usually, in multi-row matrices, diodes are combined by 8 or even 4. That is, first there are 8 pixels of the first row, then 8 of the second. Moreover, the directions of traversal of pixels in these sets of 8 can be different, say the first 8 are from right to left, the next ones are from left to right, and so on

tom3f, 2022-08-15:
> Good job, is it 160x20 internally? Did you modify the library?

The resolution is 80x40 pixel. 
No, i'm not using this library. 

To understand how such a display (it's my first :blush:) works I've painted a (incomplete) schematic : [P4_80x40_10S.pdf](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/files/9336035/P4_80x40_10S.pdf)
Maybe it helps to adapt the library.

tom3f, 2022-08-15:
> is it 160x20 internally?
 
The resolution is 80 x 40.

> Did you modify the library?

No, i'm not using this library.

To understand how such a display works (it's my first 😊) I've painted a (incomplete) schematic : [P4_80x40_10S.pdf](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/files/9336127/P4_80x40_10S.pdf)
Maybe it helps to adapt the library.

board707, 2022-08-15:
Thank you @tom3f 
As I see in the schematics - note the inscriptions on the outputs of driver UR1 - a zigzag scan of 8 pixels is used here. Each driver outputs 8 pixels in one row with indices from 0 to 7 first, then 8 in the row below in reverse order - from 7 to 0

closed 2022-08-16 by tom3f