mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #309: Problem with 64 x 64 display (striped images)

flasher69, 2022-08-15:
Hi guys,

I have big issues with the images shown on my HUB75 64 x 64 display. No matter which example I tried, I get the same results. The image shown on display is always striped in 4 sections. I've even tried to change some settings in ESP32-HUB75-MatrixPanel-I2S-DMA.h (such as MATRIX_WIDTH/HEIGHT) but the results always stay the same. I tried another panel from same brand and different ESP32 boards but always the same results.
Any idea what might be the problem?
Thank you for the help.
cheers

[image]
.

board707, 2022-08-15:
Do you connect a E pin?

ergindemir, 2022-08-15:
Some examples modify e pin afterwards. 
Modify the following line if it's in your code:
`mxconfig.gpio.e = 18;`

flasher69, 2022-08-15:
> 

Thanx a lot ergindemir. I've changed E-Pin in HUB75-MatrixPanel-I2S-DMA.h and scatches to pin 18 and now everything works :) Thank you for your help

closed 2022-08-16 by mrfaptastic