mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #312: 1/4 Scan Panel, Made of type of panel working but the other is not working

ndavi, 2022-08-19:
Hello everyone ! 

I'm trying to map some 1/4 16x32 pixels panels, i've successfully mapped one of these panels (Chaining is not working) that I bought on aliexpress, and i'm trying to map other panels that I have at home.

For common understanding, the 16x32 panel that is working will be called "**PANEL1**" and the no working panel will be called "**PANEL2**"

I've made the **PANEL1** working by using the ONE_EIGHT_32 mode, and by using a function in the virtual matrix to remap the led to the correct position, and removed the "stupidity check" : /*if ( (vmodule_rows == 1) && (vmodule_cols == 1)) // single panel...
(Yes I don't know what i'm doing)

I also set 
mxconfig.latch_blanking = 4;
mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_8M;

So, for my first test I have an esp32 connected directly to the HUB75 connector, driving all these pins by 3.3v logic and the first panel is working fine with clkphase = true, and when I connect the other panel by chaning it, it's working well also (Don't understand why)

You can see these panels working in this video :

https://user-images.githubusercontent.com/7792469/185676738-be68dbef-ddee-408e-843b-6c445f00fead.mp4

When I connect directly the **PANEL2** to the esp32, nothing seems to works fine, I have glitch and other stuff like this.

Now, i've made a PCB using 74HCT245N to up my logic to 5V, and i'm trying to do the same test.

First test with only **PANEL1** connected : The led are not properly mapped as you can see 

https://user-images.githubusercontent.com/7792469/185679759-88d55054-409d-4683-b445-b5e4e0cff3ac.mp4

Second test :  I'm connecting the **PANEL2** to **PANEL1** with the HUB75 connector, the mapping starting to work on the first panel and the second panel doesn't work as you can see

https://user-images.githubusercontent.com/7792469/185679781-18324086-7803-4bfd-b7f1-82e9a93d528f.mp4

For the **PANEL1** to work normally alone, I have to set the clkphase to false, and it's works nice without PANEL 2 connected

If i connect the **PANEL2**, the **PANEL1** works normally and the **PANEL2** is more glitchy

Now, i'm trying to connect the **PANEL2** directly to the esp32 PCB

First, I have to reset the power on the panel because some led stay on when I connect all the cables at first

Now, if I try to power only one pixel, I'm stuck with this line of pixels that stay on

I'm trying to power this pixel : 

[image]

And on the panel, this is the result :

[image]

When I try to power one entire line, I have two lines that stay on

Also, when i send like 165 values in all rgb channels, I'm stuck with multi colours line like this

[image]

Here a video when I'm trying to send a line from the top to the bottom and the from the left to the right

https://user-images.githubusercontent.com/7792469/185680278-67435df1-fbd1-4be6-ac82-9d4be2cf0003.mp4

Here a gist of what i've done, I know it's crap because it's only for some kind of testing
https://gist.github.com/ndavi/6ce8f8a4ad13ecb05b90fff316b90c0d

If anybody have any idea of what is going on, i'm happy to hear it !

The biggest chip on the back of my panel is a CN74HC138
Here a picture of the back of the panel

[image]

Thanks for your help !

board707, 2022-08-19:
> by using a function in the virtual matrix to remap the led to the correct position

Hi
What is remap function are you using?
There are few simple variants of 32x16 1/4 panels (the other more rare). Al what you need - choose the right function for remap.

ndavi, 2022-08-19:
Hello ! Thanks for answering me

The remap function that I using is one made by myself, i've not seen other functions in the library to remap 1/4 panels, and the real problem is that I can even light on a single pixel, so I think that first it's not a mapping problem at this side, and I cannot have proper colors also. I'm really not at expert at this point, so maybe it the remap function, but where can I find these rempa functions ?

board707, 2022-08-19:
> where can I find these remap functions ?

You can search it on the internet, but you can write it yourself, as you did with the first matrix. Can you show the function you used in first case?
But indeed, it may be not a map problem, if the panel used some non-standard drivers or mux

ndavi, 2022-08-19:
The function used is in this gist : 

https://gist.github.com/ndavi/6ce8f8a4ad13ecb05b90fff316b90c0d

I create a matrix with the correct mapping using these two functions 

```
inline void VirtualMatrixPanel::calculateOneLine(int16_t (&threedim)[16][32], int startColumn, int startRow, int &numLed) {
    for (int u = 0; u < 2; ++u) {
        for (int k = 0; k < 8; ++k) {
            threedim[startRow][startColumn] = numLed;
            numLed++;
            startColumn--;
        }
        startColumn++;
        for (int k = 0; k < 8; ++k) {
            threedim[startRow - 4][startColumn] = numLed;
            numLed++;
            startColumn++;
        }
        startColumn = startColumn - 9;
    }
}

inline void VirtualMatrixPanel::calculateOneSide(int16_t (&threedim)[16][32], int startColumn, int startRow, int &numLed) {
    for (int i = 0; i < 2; ++i) {
        for (int j = 0; j < 4; ++j) {
            calculateOneLine(threedim, startColumn, startRow, numLed);
            startRow++;
        }
        startRow = startRow + 4;
    }
}
```

The method is called like this : 
```
        // Calculate the matrix to map the one quart panel
        int numLed = 0;
        int startRow = 4;
        int startColumn = 31;
        calculateOneSide(oneQuartPanelMatrix, startColumn, startRow, numLed);
        startColumn = 15;
        calculateOneSide(oneQuartPanelMatrix, startColumn, startRow, numLed);
```

And with the created matrix, i map the good coordinate like this : 

```
    //Remap
    int16_t n = oneQuartPanelMatrix[y][x];
    x = n % 32;
    y = n / 32;
```

I think this is the good mapping for this panel, because like I said when I'm using the "PANEL1" on my first example, everything seems to works fine, with the "PANEL2" (The panel that i want to map correctly) so it's a strange behaviour.. and it's worked only when my logic level was at 3.3v, now that i'm at 5v I can't have the "PANEL2" working properly even if I using the "PANEL1" at the first panel in my chain, but the "PANEL1" is working fine

board707, 2022-08-19:
It is cool really, thank you

But this approach looks too complicated to rebuild it to other scanning patterns.
Perhaps I will try to simplify this function

ndavi, 2022-08-20:
Thanks !

I started to made some things less crap by reducing the brightness to 128, and using a latch_blanking to 30, but it's not helping my problem as I need these led to be at full power and a lot of glitch are remaining

Here is a video with a line going from left to right

https://user-images.githubusercontent.com/7792469/185722145-55735089-d914-4d4e-9de4-f25d57fb64a0.mp4

So I starting to think that maybe the problem is related to clock or something that is related to timings

board707, 2022-08-20:
If you have a Raspberry Pi - you can test your matrix with [hzeller's library](https://github.com/hzeller/rpi-rgb-led-matrix), which has a lot of matrix types and scan options. If some set of settings allows you to run the matrix, it will be easier to port it here later.
Or if you have STM32 bluepill or blackpill boards - you can try the [DMD_STM32](https://github.com/board707/DMD_STM32) library

mrfaptastic, 2022-08-20:
As per what be@board707 said. Also try panel2 using PxMatrix or something, and see if that works.

Electrically it seems these boards are different right? Are they both the EXACT same chips?

It is going to be really painful to try and get this to work properly with chaining types of chips, as you will probably never be able to get both to work perfect when together.

PANEL2 doesn't like the clock speed or something, has some weird chip like FM chip that needs register settings to be loaded before use, or has a fault with it, I think.

ndavi, 2022-08-20:
Thanks to both of you !

> Electrically it seems these boards are different right? They need to both be the EXACT same chips.

Yes it's different board, but I don't want to make them works chained together, i'm just trying things with the stuff I have at home

I will try to make these works with other matrix library !

ndavi, 2022-08-20:
Ok so I made it kind of works using the SmartMatrix library, now I seems to experiments some type of lite ghosting with "PANEL1" and "PANEL2", I will ask for help on the pixelmatic forums as it's not related to this library now

Thanks for your help

closed 2022-08-20 by ndavi