mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #313: Gif animation and "pure" black color.

GrimorJander, 2022-08-27:
Hello everyone. Just commenting on something I have noticed while playing gif images. Both those pixels that should be pure black or transparent are slightly lit, so the perception of contrast is reduced. It will also consume a little more energy and reduce its useful life if those pixels never rest completely off. 
This only occurs in the gif player, not in other examples of text, patterns or still image pixel-by-pixel mapping.
Does anyone know anything about this effect?

mrfaptastic, 2022-08-28:
You'd have to look at the colour palette of the gif and see if black is actually black. #000

GrimorJander, 2022-08-28:
> You'd have to look at/edit the colour palette of the gif and see if black is actually black. #000
> 
> Otherwise if it's like some low values for RG and B then it's probably enough to trigger the pixels.

I bet it's the second. because it happens to me in all blacks and transparents, so it's not from the color palette.

Could a resistor lower that voltage to see if it improves the pixel blackout?

Thanks again Mr.

mrfaptastic, 2022-08-28:
> I bet it's the second. because it happens to me in all blacks and transparents, so it's not from the color palette.

Interesting. Is that when only using the GIF library, or not? If there is always a glow with the panel then there's some other electrical issue.

> Could a resistor lower that voltage to see if it improves the pixel blackout?

These panels work on a digital signal. You don't want to put a resister across the input pins.

> Thanks again Mr.

Anytime. I hope you like the library.

GrimorJander, 2022-08-28:
Really apreciate your work with this library. By far, the best for rgb matrix panel.

GrimorJander, 2022-08-31:
I just noticed that at the beginning in the first round of playback those black pixels look perfectly black (pixel off) and it is with the loading of the following ones that the screen remains excited and they do not finish turning off. With what are not the images or the panel/hardware. It's probably something I've broken in the code.
I've tried putting black fills or screen wipes (`dma_display->clearScreen(); dma_display->fillScreen(dma_display->color565(0, 0, 0));`) in the loop but it doesn't improve the behavior.

GrimorJander, 2022-08-31:
If i plug the esp32 to power first, and with program running then  power the matrix panel blacks are perfect "pure black" but if i reset the module or restart with matrix panel powered, blacks are "little dim grey"
[image]
[image]

ergindemir, 2022-09-01:
To conclude that it's not a library issue, you have to test it with another library.

mrfaptastic, 2022-09-01:
Have you tried playing with the clkphase configuration value?

Looks like some random electrical issue to me. Check your grounding, shorten your cables, check the voltage etc.

GrimorJander, 2022-09-09:
> Have you tried playing with the clkphase configuration value?
> 
> Looks like some random electrical issue to me. Check your grounding, shorten your cables, check the voltage etc.

That was the thing!

Changing the setting to `mxconfig.clkphase = true;` solves my problem.

Thank you very much for your support.

GrimorJander, 2022-09-09:
Consider to ad to the "Common Issues Guide" to help others with Gif Animation example.

closed 2022-09-09 by GrimorJander