mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #317: 64x32 1/16 panel not displaying on lines 9-16 and 25-32 - Wrapped back

vanceb, 2022-09-09:
Hi I have a 64x32 1/16 scan panel and it is displaying line 9 at line 1, 10 at 2, etc  Similar for bottom half of the display.  See [video](https://user-images.githubusercontent.com/9115601/189382878-d0026f52-b5c0-440a-a69f-622b13ad0677.mp4)

It relates to the D pin as disconnecting this cable makes no difference to the display.

I have:
* Replaced the cable that connects the D pin
* Double-checked the software pin assignment matches the physical pin

Excerpt from the sketch:

~~~c++
//Pinout
#define R1_PIN  (25)
#define G1_PIN  (26)
#define B1_PIN  (27)
#define R2_PIN  (12)
#define G2_PIN  (13)
#define B2_PIN  (14)
#define CLK_PIN (19)
#define LAT_PIN (32)
#define OE_PIN  (33)
#define A_PIN   (16)
#define B_PIN   (15)
#define C_PIN   (17)
#define D_PIN   (18)
#define E_PIN   (-1)

  HUB75_I2S_CFG::i2s_pins _pins={R1_PIN, 
                                 G1_PIN, 
                                 B1_PIN, 
                                 R2_PIN, 
                                 G2_PIN, 
                                 B2_PIN, 
                                 A_PIN, 
                                 B_PIN, 
                                 C_PIN, 
                                 D_PIN, 
                                 E_PIN, 
                                 LAT_PIN, 
                                 OE_PIN, 
                                 CLK_PIN
                                 };

  // Module configuration
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X,   // module width
    PANEL_RES_Y,   // module height
    PANEL_CHAIN,   // Chain length
    _pins          // Custom pins
  );
~~~

What am I missing?

board707, 2022-09-09:
If this is single panel - perhaps it have D-pin damaged/disconnected in PCB.
Other option that you have 1/8scan panel rather than 1/16. What are the markings on the back of the panel?

vanceb, 2022-09-10:
[image]

The only real markings on the back of the panel are **P4-256x128-2121-A5**, not sure where the 256x128 comes from as the panel is definitely 64x32 LEDs.

I have had this panel running using the [NeoCat library](https://github.com/NeoCat/ESP32-P3RGB64x32MatrixPanel), but was getting flicker on update.  I will go back to using that library just to test that the wiring and panel are not broken.  If it works using that library then I will let you know.

ergindemir, 2022-09-10:
It's 16s because there are 24 chips.

vanceb, 2022-09-10:
[image]

Confirm that panel and wiring are OK - This is the output when using the NeoCat library.

Any further suggestions?

vanceb, 2022-09-10:
OK, Solved - my mistake when mixing the examples.  Altough I had set the pins as shown in my code excerpts above, slightly further down in the code was this line:

~~~c++
mxconfig.gpio.e = 18;
~~~

Which overrides the earlier declaration in `_pins` where the `E_PIN` was set to -1.  For completeness my code is as follows:

~~~c++
#include <Arduino.h>

//Pinout
#define R1_PIN  (25)
#define G1_PIN  (26)
#define B1_PIN  (27)
#define R2_PIN  (12)
#define G2_PIN  (13)
#define B2_PIN  (14)
#define CLK_PIN (19)
#define LAT_PIN (32)
#define OE_PIN  (33)
#define A_PIN   (16)
#define B_PIN   (15)
#define C_PIN   (17)
#define D_PIN   (18)
#define E_PIN   (-1)

#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

#define PANEL_RES_X 64      // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32     // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1      // Total number of panels chained one to another
 
//MatrixPanel_I2S_DMA dma_display;
MatrixPanel_I2S_DMA *dma_display = nullptr;

uint16_t myBLACK = dma_display->color565(0, 0, 0);
uint16_t myWHITE = dma_display->color565(255, 255, 255);
uint16_t myRED = dma_display->color565(255, 0, 0);
uint16_t myGREEN = dma_display->color565(0, 255, 0);
uint16_t myBLUE = dma_display->color565(0, 0, 255);

// Input a value 0 to 255 to get a color value.
// The colours are a transition r - g - b - back to r.
// From: https://gist.github.com/davidegironi/3144efdc6d67e5df55438cc3cba613c8
uint16_t colorWheel(uint8_t pos) {
  if(pos < 85) {
    return dma_display->color565(pos * 3, 255 - pos * 3, 0);
  } else if(pos < 170) {
    pos -= 85;
    return dma_display->color565(255 - pos * 3, 0, pos * 3);
  } else {
    pos -= 170;
    return dma_display->color565(0, pos * 3, 255 - pos * 3);
  }
}

void drawText(int colorWheelOffset)
{
  
  // draw text with a rotating colour
  dma_display->setTextSize(1);     // size 1 == 8 pixels high
  dma_display->setTextWrap(false); // Don't wrap at end of line - will do ourselves

  dma_display->setCursor(5, 0);    // start at top left, with 8 pixel of spacing
  uint8_t w = 0;
  const char *str = "ESP32 DMA";
  for (w=0; w<strlen(str); w++) {
    dma_display->setTextColor(colorWheel((w*32)+colorWheelOffset));
    dma_display->print(str[w]);
[... 109 lines cut ...]
uint8_t wheelval = 0;
void loop() {

    // animate by going through the colour wheel for the first two lines
    drawText(wheelval);
    wheelval +=1;

    delay(20); 
/*
  drawText(0);
  delay(2000);
  dma_display->clearScreen();
  dma_display->fillScreen(myBLACK);
  delay(2000);
  dma_display->fillScreen(myBLUE);
  delay(2000);
  dma_display->fillScreen(myRED);
  delay(2000);
  dma_display->fillScreen(myGREEN);
  delay(2000);
  dma_display->fillScreen(myWHITE);
  dma_display->clearScreen();
  */
}
~~~

closed 2022-09-10 by vanceb