mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #323: Question about P3 : 64*64 1/32

hasanedrees25, 2022-09-27:
does the library support the panel with the driver  ICN2037BP ?

DarrylStrong, 2022-09-27:
Looks to me like you have your D line missing....

hasanedrees25, 2022-09-27:
@DarrylStrong  yes my pannel dont have it 
[image]

DarrylStrong, 2022-09-27:
There is something odd with the markings as it has an E...

Check to see which pins are actually ground as I suspect the standard D pin is marked as GND.

mzashh, 2023-04-19:
i have the same panel and the D pin in next to C and is for some reason labelled as ground, it works with the library without any modifications

DPigeon, 2024-01-27:
> i have the same panel and the D pin is next to the C pin and is for some reason labelled as ground, it works with the library without any modifications
> 
> [image]
> 
> 
> 
> 

@mzashh I have the same panel with same chip except mine is a bit newer. So you did not add any specific config to your code, right? I get no output display on my panel or gargabe lines.