mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #335: Question about P3: 64*64 1/32 Chained panel to 2 cols and 2 rows

secangkirkopipanas, 2022-10-22:
Hi @mrfaptastic,

I'm currently facing issue with my P3 64x64 chained panel. I'm making 2 rows by 3 cols chained panels.
When I set NUM_ROWS 1 and NUM_COLS 3, it worked with panel 4-6 cloned the output of panel 1-3.
But when I set NUM_ROWS 2 and NUM_COLS 3, it's just blank without any output to all panels.

I have set the MATRIX_HEIGHT to 64 in ESP32-HUB75-MatrixPanel-I2S-DMA.h.
Appreciate your assistance. Thank you.

**Pin configuration:**
R1_PIN  15
G1_PIN  13
B1_PIN   2
R2_PIN  4
G2_PIN  14
B2_PIN  16
A_PIN  17
B_PIN  26
C_PIN  5
D_PIN  25
E_PIN  1
LAT_PIN  33
OE_PIN  19
CLK_PIN  18

**Some lines of code:**
HUB75_I2S_CFG mxconfig(
  PANEL_RES_X,    // module width
  PANEL_RES_Y,    // module height
  PANEL_CHAIN     // Chain length
);
dmaDisplay = new MatrixPanel_I2S_DMA(mxconfig);
dmaDisplay->begin(R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN);
vDisplay = new VirtualMatrixPanel((*dmaDisplay), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, SERPENT, TOPDOWN);

mrfaptastic, 2022-10-22:
What's the serial debug output when running with 6 panels? Probably out of memory....

mrfaptastic, 2022-10-22:
What's the serial debug output when running with 6 panels? Probably out of memory.

secangkirkopipanas, 2022-10-22:
This is the serial debug output:

```
19:56:08.099 -> ets J⸮)A 8⸮⸮b⸮⸮_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:56:08.099 -> mode:DIO, clock div:2
19:56:08.099 -> load:0x3fff0030,len:1184
19:56:08.099 -> load:0x40078000,len:13132
19:56:08.099 -> load:0x40080400,len:3036
19:56:08.133 -> entry 0x400805e4
19:56:08.781 -> [ ⸮⸮mum⸮⸮⸮⸮2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
19:56:08.818 -> 
19:56:08.818 -> ###############################################
19:56:08.818 -> # Welcome to Smart Mini 4WD RGB Timer         #
19:56:08.818 -> # ------------------------------------------- #
19:56:08.818 -> # By NgebutBenjut Mini 4WD                    #
19:56:08.818 -> ###############################################
19:56:08.818 -> 
19:56:08.818 -> Initialising data in SPIFFS...
19:56:08.919 -> [   121][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
19:56:08.919 -> [   122][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
19:56:08.919 -> [   127][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
19:56:08.919 -> [   127][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
19:56:08.919 -> [   133][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
19:56:08.919 -> [   137][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
19:56:08.919 -> Configuration file is loaded successfully.
19:56:08.954 -> LOG LEVEL: 1
19:56:08.954 -> OTA MODE: 0
19:56:08.954 -> [INFO]  Initialising BLE...
19:56:09.642 -> [   871][V][BLEDevice.cpp:76] createServer(): >> createServer
19:56:09.642 -> [   872][V][BLEServer.cpp:290] registerApp(): >> registerApp - 0
19:56:09.679 -> [   872][V][FreeRTOS.cpp:189] take(): Semaphore taking: name: RegisterAppEvt (0x3ffd7370), owner: <N/A> for registerApp
19:56:09.679 -> [   882][V][FreeRTOS.cpp:198] take(): Semaphore taken:  name: RegisterAppEvt (0x3ffd7370), owner: registerApp
19:56:09.679 -> [   891][V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: RegisterAppEvt (0x3ffd7370), owner: registerApp for registerApp
19:56:09.679 -> [   893][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_REG_EVT
19:56:09.714 -> [   914][V][BLEUtils.cpp:1530] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_REG_EVT
19:56:09.714 -> [   921][V][BLEUtils.cpp:1673] dumpGattServerEvent(): [status: ESP_GATT_OK, app_id: 0]
19:56:09.714 -> [   929][V][BLEServer.cpp:144] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_REG_EVT
19:56:09.714 -> [   938][V][FreeRTOS.cpp:143] give(): Semaphore giving: name: RegisterAppEvt (0x3ffd7370), owner: registerApp
19:56:09.750 -> [   948][V][BLEServer.cpp:280] handleGATTServerEvent(): << handleGATTServerEvent
19:56:09.750 -> [   948][V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: RegisterAppEvt (0x3ffd7370), owner: <N/A>
19:56:09.750 -> [   965][V][BLEServer.cpp:294] registerApp(): << registerApp
19:56:09.750 -> [   970][V][BLEDevice.cpp:83] createServer(): << createServer
19:56:09.750 -> [   976][V][BLEServer.cpp:67] createService(): >> createService - 4fbfc284-1fb5-459e-8fdc-c9d9c331924d
19:56:09.784 -> [   984][V][FreeRTOS.cpp:189] take(): Semaphore taking: name: CreateEvt (0x3ffe3a24), owner: <N/A> for createService
19:56:09.784 -> [   995][V][FreeRTOS.cpp:198] take(): Semaphore taken:  name: CreateEvt (0x3ffe3a24), owner: createService
19:56:09.784 -> [  1005][V][BLEService.cpp:60] executeCreate(): >> executeCreate() - Creating service (esp_ble_gatts_create_service) service uuid: 4fbfc284-1fb5-459e-8fdc-c9d9c331924d
19:56:09.822 -> [  1019][V][FreeRTOS.cpp:189] take(): Semaphore taking: name: CreateEvt (0x3ffe3c90), owner: <N/A> for executeCreate
19:56:09.822 -> [  1029][V][FreeRTOS.cpp:198] take(): Semaphore taken:  name: CreateEvt (0x3ffe3c90), owner: executeCreate
19:56:09.822 -> [  1039][V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: CreateEvt (0x3ffe3c90), owner: executeCreate for executeCreate
19:56:09.856 -> [  1040][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CREATE_EVT
19:56:09.856 -> [  1061][V][BLEUtils.cpp:1530] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CREATE_EVT
19:56:09.856 -> [  1069][V][BLEUtils.cpp:1599] dumpGattServerEvent(): [status: ESP_GATT_OK, service_handle: 40 0x28, service_id: [uuid: 4fbfc284-1fb5-459e-8fdc-c9d9c331924d, inst_id: 0]]
19:56:09.889 -> [  1084][V][BLEServer.cpp:144] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CREATE_EVT
19:56:09.889 -> [  1093][V][FreeRTOS.cpp:143] give(): Semaphore giving: name: CreateEvt (0x3ffe3a24), owner: createService
19:56:09.889 -> [  1102][V][BLEService.cpp:194] setHandle(): >> setHandle - Handle=0x28, service UUID=4fbfc284-1fb5-459e-8fdc-c9d9c331924d)
19:56:09.889 -> [  1113][V][BLEService.cpp:200] setHandle(): << setHandle
19:56:09.924 -> [  1118][V][FreeRTOS.cpp:143] give(): Semaphore giving: name: CreateEvt (0x3ffe3c90), owner: executeCreate
19:56:09.924 -> [  1128][V][BLEServer.cpp:280] handleGATTServerEvent(): << handleGATTServerEvent
19:56:09.924 -> [  1128][V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: CreateEvt (0x3ffe3c90), owner: <N/A>
19:56:09.924 -> [  1144][V][BLEService.cpp:76] executeCreate(): << executeCreate
19:56:09.958 -> [  1150][V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: CreateEvt (0x3ffe3a24), owner: <N/A> for createService
[... cut ...]
19:56:15.332 -> E (6550) ledc: freq_hz=0 duty_resolution=10
19:56:15.332 -> [  6555][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
19:56:15.367 -> [INFO]  RGB MATRIX details
19:56:15.367 -> [INFO]  MATRIX_WIDTH  : 256
19:56:15.367 -> [INFO]  MATRIX_HEIGHT : 64
19:56:15.367 -> E (6570) ledc: freq_hz=0 duty_resolution=10
19:56:15.367 -> [  6575][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
19:56:15.367 -> [INFO]  ---
19:56:15.367 -> [INFO]  Ready to use!
```

Is there anything you can catch? Thank you.

mrfaptastic, 2022-10-22:
Can you compile your project with      -DCORE_DEBUG_LEVEL=2 please

If using platform IO, try something like
```
[common_env_data]
build_flags = 
     -DSERIAL_DEBUG=1
     -DCORE_DEBUG_LEVEL=2

```

secangkirkopipanas, 2022-10-23:
Hi @mrfaptastic,

My previous log was with core debug level "Verbose".

Following is the log with core debug level "Debug", please ignore the ledc error, still trying to fix this...
```
11:18:43.201 -> ⸮⸮ ⸮ ⸮⸮⸮ ⸮   ⸮ ⸮⸮⸮⸮⸮ ⸮⸮  ⸮Dy⸮$w:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
11:18:43.201 -> mode:DIO, clock div:2
11:18:43.201 -> load:0x3fff0030,len:1344
11:18:43.201 -> load:0x40078000,len:13836
11:18:43.201 -> load:0x40080400,len:3608
11:18:43.201 -> entry 0x400805f0
11:18:43.793 -> [ ⸮⸮mum⸮⸮⸮⸮2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
11:18:43.793 -> 
11:18:43.793 -> ###############################################
11:18:43.793 -> # Welcome to Smart Mini 4WD RGB Timer         #
11:18:43.793 -> # ------------------------------------------- #
11:18:43.793 -> # By NgebutBenjut Mini 4WD                    #
11:18:43.793 -> ###############################################
11:18:43.826 -> 
11:18:43.826 -> Initialising data in SPIFFS...
11:18:43.826 -> [    53][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
11:18:43.826 -> [    53][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
11:18:43.826 -> [    57][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
11:18:43.826 -> [    58][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
11:18:43.860 -> [    64][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
11:18:43.860 -> [    68][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
11:18:43.860 -> Configuration file is loaded successfully.
11:18:43.860 -> LOG LEVEL: 1
11:18:43.860 -> OTA MODE: 0
11:18:43.860 -> [INFO]  Initialising BLE...
11:18:44.560 -> [   795][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_REG_EVT
11:18:44.597 -> [   797][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CREATE_EVT
11:18:44.597 -> [   806][D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=bec5483e-33a1-4698-b7f5-ab08364b26a8 to service: UUID: 4fbfc284-1fb5-459e-8fdc-c9d9c331924d, handle: 0x0028
11:18:44.597 -> [   823][D][BLECharacteristic.cpp:90] executeCreate(): Registering characteristic (esp_ble_gatts_add_char): uuid: bec5483e-33a1-4698-b7f5-ab08364b26a8, service: UUID: 4fbfc284-1fb5-459e-8fdc-c9d9c331924d, handle: 0x0028
11:18:44.634 -> [   842][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_ADD_CHAR_EVT
11:18:44.634 -> [   853][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_ADD_CHAR_DESCR_EVT
11:18:44.634 -> [   864][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_START_EVT
11:18:44.667 -> [   874][I][BLEDevice.cpp:577] getAdvertising(): create advertising
11:18:44.667 -> [   880][D][BLEDevice.cpp:579] getAdvertising(): get advertising
11:18:44.667 -> [   886][D][BLEDevice.cpp:579] getAdvertising(): get advertising
11:18:44.667 -> [   891][D][BLEAdvertising.cpp:199] start(): - advertising service: 4fbfc284-1fb5-459e-8fdc-c9d9c331924d
11:18:44.667 -> [INFO]  BLE was started successfully![   905][D][BLEDevice.cpp:579] getAdvertising(): get advertising
11:18:44.700 -> 
11:18:44.700 -> [   914][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 0]
11:18:44.700 -> [INFO]  - Characteristic UUID: bec5483e-33a1-4698-b7f5-ab08364b26a8
11:18:44.700 -> [   917][D][BLEDevice.cpp:579] getAdvertising(): get advertising
11:18:44.700 -> [INFO]  - Service UUID: 4fbfc284-1fb5-459e-8fdc-c9d9c331924d
11:18:44.738 -> [   929][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 1]
11:18:44.738 -> [   940][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
11:18:44.738 -> [   941][D][BLEDevice.cpp:579] getAdvertising(): get advertising
11:18:44.738 -> [   947][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
11:18:44.738 -> [   952][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 6]
11:18:45.401 -> E (1617) ledc: freq_hz=0 duty_resolution=10
11:18:45.401 -> [  1621][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
11:18:45.543 -> E (1754) ledc: freq_hz=0 duty_resolution=10
11:18:45.543 -> [  1757][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
11:18:45.650 -> E (1890) ledc: freq_hz=0 duty_resolution=10
11:18:45.688 -> [  1893][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
11:18:45.797 -> E (2026) ledc: freq_hz=0 duty_resolution=10
11:18:45.797 -> [  2029][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
11:18:45.869 -> E (2094) ledc: freq_hz=0 duty_resolution=10
11:18:45.869 -> [  2097][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
11:18:46.088 -> E (2298) ledc: freq_hz=0 duty_resolution=10
11:18:46.088 -> [  2301][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
11:18:46.343 -> E (2570) ledc: freq_hz=0 duty_resolution=10
11:18:46.343 -> [  2573][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
11:18:46.485 -> E (2706) ledc: freq_hz=0 duty_resolution=10
11:18:46.485 -> [  2709][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
11:18:46.630 -> E (2842) ledc: freq_hz=0 duty_resolution=10
[... 29 lines cut ...]
11:18:48.714 -> E (4950) ledc: freq_hz=0 duty_resolution=10
11:18:48.714 -> [  4953][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
11:18:48.787 -> E (5018) ledc: freq_hz=0 duty_resolution=10
[... cut ...]
11:18:49.217 -> [  5429][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
11:18:49.468 -> E (5698) ledc: freq_hz=0 duty_resolution=10
11:18:49.468 -> [  5701][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
11:18:49.543 -> [INFO]  RGB MATRIX details
11:18:49.543 -> [INFO]  MATRIX_WIDTH  : 192
11:18:49.543 -> [INFO]  MATRIX_HEIGHT : 128
11:18:49.543 -> E (5751) ledc: freq_hz=0 duty_resolution=10
11:18:49.543 -> [  5754][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
11:18:49.577 -> [INFO]  ---
11:18:49.577 -> [INFO]  Ready to use!

```

mrfaptastic, 2022-10-23:
Sorry. I should have been more specific - are you using the git version of this library, or the last published 2.0.7 version? If you could use the latest git version - and provide the serial output.

secangkirkopipanas, 2022-10-23:
Hi, I'm currently using v2.0.7 version from the release. Let me try to use the latest git version and get the serial output.

secangkirkopipanas, 2022-10-24:
Hi @mrfaptastic,

After applying the latest changes from Github, I got the following error in the log:
```
18:59:55.299 -> [  1944][E][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:32] allocateDMAmemory(): [MatrixPanel] CRITICAL ERROR: Can't allocate rowBitStruct 26! Not enough memory for requested PIXEL_COLOUR_DEPTH_BITS. Please reduce PIXEL_COLOUR_DEPTH_BITS value.
```

Full log:
```
18:59:52.813 -> ⸮⸮ ⸮ ⸮⸮⸮ A⸮ ⸮0k⸮E⸮⸮⸮0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
18:59:52.813 -> mode:DIO, clock div:1
18:59:52.813 -> load:0x3fff0030,len:1344
18:59:52.813 -> load:0x40078000,len:13864
18:59:52.855 -> load:0x40080400,len:3608
18:59:52.855 -> entry 0x400805f0
18:59:53.328 -> [ ⸮⸮mum⸮⸮⸮⸮2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
18:59:53.328 -> E (376) psram: PSRAM ID read error: 0xffffffff
18:59:53.362 -> [     6][W][esp32-hal-psram.c:71] psramInit(): PSRAM init 
18:59:53.362 -> ###############################################
18:59:53.362 -> # Welcome to Smart Mini 4WD RGB Timer         #
18:59:53.362 -> # ------------------------------------------- #
18:59:53.362 -> # By NgebutBenjut Mini 4WD                    #
18:59:53.362 -> ###############################################
18:59:53.362 -> 
18:59:53.362 -> Initialising data in SPIFFS...
18:59:53.399 -> [    58][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
18:59:53.399 -> [    58][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
18:59:53.399 -> [    60][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
18:59:53.399 -> [    63][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
18:59:53.399 -> [    69][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
18:59:53.399 -> [    73][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
18:59:53.432 -> OTA MODE: 0
18:59:53.432 -> Bluetooth is enabled!
18:59:53.432 -> [    91][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
18:59:53.432 -> [    91][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
18:59:55.148 -> [  1821][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:345] begin(): [begin()] Using GPIO 15 for R1_PIN
18:59:55.148 -> [  1821][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:346] begin(): [begin()] Using GPIO 13 for G1_PIN
18:59:55.148 -> [  1826][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:347] begin(): [begin()] Using GPIO 2 for B1_PIN
18:59:55.186 -> [  1834][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:348] begin(): [begin()] Using GPIO 4 for R2_PIN
18:59:55.186 -> [  1843][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:349] begin(): [begin()] Using GPIO 14 for G2_PIN
18:59:55.186 -> [  1851][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:350] begin(): [begin()] Using GPIO 16 for B2_PIN
18:59:55.186 -> [  1859][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:351] begin(): [begin()] Using GPIO 17 for A_PIN
18:59:55.224 -> [  1868][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:352] begin(): [begin()] Using GPIO 26 for B_PIN
18:59:55.224 -> [  1876][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:353] begin(): [begin()] Using GPIO 5 for C_PIN
18:59:55.224 -> [  1884][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:354] begin(): [begin()] Using GPIO 25 for D_PIN
18:59:55.224 -> [  1892][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:355] begin(): [begin()] Using GPIO 1 for E_PIN
18:59:55.224 -> [  1900][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:356] begin(): [begin()] Using GPIO 33 for LAT_PIN
18:59:55.261 -> [  1909][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:357] begin(): [begin()] Using GPIO 19 for OE_PIN
18:59:55.261 -> [  1917][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:358] begin(): [begin()] Using GPIO 18 for CLK_PIN
18:59:55.261 -> [  1926][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:16] allocateDMAmemory(): [MatrixPanel] Free heap: 178088
18:59:55.261 -> [  1935][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:17] allocateDMAmemory(): [MatrixPanel] Free SPIRAM: 0
18:59:55.299 -> [  1944][E][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:32] allocateDMAmemory(): [MatrixPanel] CRITICAL ERROR: Can't allocate rowBitStruct 26! Not enough memory for requested PIXEL_COLOUR_DEPTH_BITS. Please reduce PIXEL_COLOUR_DEPTH_BITS value.
18:59:55.299 -> 
18:59:55.794 -> Successfully opened GIF; Canvas size = 192 x 64
19:00:03.935 -> E (10981) ledc: freq_hz=0 duty_resolution=10
19:00:03.935 -> [ 10609][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
19:00:04.085 -> E (11117) ledc: freq_hz=0 duty_resolution=10
19:00:04.085 -> [ 10745][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
19:00:04.195 -> E (11253) ledc: freq_hz=0 duty_resolution=10
19:00:04.233 -> [ 10881][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
19:00:04.346 -> E (11389) ledc: freq_hz=0 duty_resolution=10
19:00:04.346 -> [ 11017][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
19:00:04.416 -> E (11457) ledc: freq_hz=0 duty_resolution=10
19:00:04.416 -> [ 11085][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
19:00:04.632 -> E (11661) ledc: freq_hz=0 duty_resolution=10
19:00:04.632 -> [ 11289][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
19:00:04.888 -> E (11933) ledc: freq_hz=0 duty_resolution=10
19:00:04.888 -> [ 11561][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
19:00:05.034 -> E (12069) ledc: freq_hz=0 duty_resolution=10
19:00:05.034 -> [ 11697][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
[... 66 lines cut ...]
19:00:08.182 -> BLE Service UUID: 
19:00:08.182 -> BLE Characteristic UUID: 
19:00:08.182 -> WIFI SSID: 
19:00:08.182 -> WIFI Password: 
19:00:08.182 -> OTA Enabled: 0
19:00:08.182 -> Timer Mode: 5
19:00:08.182 -> Sensor: laser
[... cut ...]
19:00:08.182 -> Lane 3 Enabled: 1
19:00:08.216 -> Lane 4 Enabled: 1
19:00:08.216 -> Lane 5 Enabled: 1
19:00:08.216 -> Splash 1: TAMIYA
19:00:08.216 -> Splash 2: MINI 4WD
19:00:08.216 -> * SerialDebug: Please press ? or another command and enter to activate debugs
19:00:08.216 -> * SerialDebug: Please press ? or another command and enter to activate debugs
19:00:08.216 -> * SerialDebug: Please press ? or another command and enter to activate debugs
19:00:08.216 -> * Please verify if is in Newline mode in monitor serial

```

mrfaptastic, 2022-10-24:
Yep. As expected you can't drive that many panels with that colour depth with the memory you have available.

Read: https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/blob/master/README.md#features

Change the value of PIXEL_COLOUR_DEPTH_BITS to '4' or something.

secangkirkopipanas, 2022-10-24:
Hi @mrfaptastic,

Yeah, it solved by changing PIXEL_COLOUR_DEPTH_BITS value to '6'. You can close this ticket.
Thank you for assisting.