mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #341: My red's are green, my green's are blue, and my blue's are red....

tychop, 2022-11-14:
Hi all,

Not a library issue, but looking for advice...

Im connecting a 128x64 x2 serpent setup to my esp (esp32 Trinity board).
I got everything working great, except the colors.
Like the title suggests... All my colors are shifted: My red's are green, my green's are blue, and my blue's are red....

Could it be a PIN definition issue?

These are my PIN defaults now:

```
#define A_PIN 23
#define B_PIN 19
#define C_PIN 5
#define D_PIN 17
#define E_PIN 18
#define LAT_PIN 4
#define OE_PIN 15
#define CLK_PIN 16
#define R1_PIN 25
#define R2_PIN 14
#define G1_PIN 26
#define G2_PIN 12
#define B1_PIN 27
#define B2_PIN 13
```

I tried switching the R1, R2, G1, G2, B1 and B2 around and see some colors changing, but I can't figure out the logic.....

Is there anything I can change in this pin setup to correct the colors?

mrfaptastic, 2022-11-14:
It's an issue with your panel - they've put the pins in a different order than 'normal' panels.
Swap the physical wires around or just change the GPIO mapping as you allude to above.

mrfaptastic, 2022-11-14:
Swap the physical wires around or just flip the GPIO mapping as you allude to above....
i.e. whatever was the value was of the pins, just rotate the numbers around R2 = G2, B2=G2, etc. Same for R1's

tychop, 2022-11-14:
Thnx buddy. I got it going.
I ended up with the following:

```cpp
#define B1_PIN 25
#define B2_PIN 14
#define R1_PIN 26
#define R2_PIN 12
#define G1_PIN 27
#define G2_PIN 13
```

closed 2022-11-14 by tychop