mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #348: Where do I define PIXEL_COLOUR_DEPTH_BITS ??

tychop, 2022-12-12:
I have this define basically at the first line in my program: ```#define PIXEL_COLOUR_DEPTH_BITS 7```
But it doesn't get picked up. I just get a crash (because 8 bits consumes too much mem for my program).

And when I set it to 7 in ESP32-HUB75-MatrixPanel-I2S-DMA.h on line 75, my programs works fine:
```
// 8bit per RGB color = 24 bit/per pixel,
// might be reduced to save DMA RAM
#ifndef PIXEL_COLOUR_DEPTH_BITS
 #define PIXEL_COLOUR_DEPTH_BITS      7
#endif
```

I really don't want to change stuff in the lib, so where am I supposed to define PIXEL_COLOUR_DEPTH_BITS ?

mrfaptastic, 2022-12-12:
What development environment are you using? I use PlatformIO, and then in my platformio.ini set a global define which then gets picked up everywhere when the sketch is compiled.

The scope of 'defines' is only within the .cpp/.c or .hpp/.h file to which they were created, and given Arduino doesn't have a way to declare global defines (I don't think), you'll need to edit the library as a result.

```
[env:lolin_d32]
platform = espressif32
board = lolin_d32
framework = arduino
monitor_speed = 115200
upload_speed = 921600
;board_build.filesystem = littlefs
build_flags = 
	${common_env_data.build_flags}
	-DPIXEL_COLOUR_DEPTH_BITS=7
```

tychop, 2022-12-13:
Thanks, that worked.
I'm such a noob in this language.

closed 2022-12-13 by tychop

mrfaptastic, 2022-12-13:
> Thanks, that worked.
> I'm such a noob in this language.

🙂 We all are at some point. That's why I started this library, namely to have a fun way to learn coding.