mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #350: Mirroring Text

JKBusse, 2022-12-13:
hey,
is there a way to mirror text?

Thanks

mrfaptastic, 2022-12-13:
Unless the AdafruitGFX library has that, then no. You'll need to write the logic yourself / modify drawPixel() function of that library to calculate 'mirror' coordinates and update as such.

board707, 2023-01-28:
> You'll need to modify drawPixel() function of that library

I think one will need to modify drawChar() or drawString()

xemjeff, 2023-03-20:
A few ways I can thing of to do this:

1) modify an exiting Arduino GFX font (.h file) - flip each of the characters. 
using this tool: https://tchapi.github.io/Adafruit-GFX-Font-Customiser/

2) modify an existing TTF font using Glyph Studio (better graphics tools), then convert to Arduino GFX font.

3) modify the drawChar function on the Arduino library and the  GFX font structure. You could add a bool flag for mirror, and if mirrored, render the character in mirror format.  You would also have to supply a print function (right to left) instead of left to right, assuming your are using some left to right char set to start with.

I guess is depends on how badly you want it.