mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #351: esp-idf standalone support

chegewara, 2022-12-17:
Hi,
will you accept small PR which will add esp-idf standalone support (no arduino as component)? (tested with esp-idf 5.0)
If yes, what is better way to do it:
- to replace all `digitalWrite` with esp-idf API (its cleaner), or 
- to add this snippet:
```
#ifdef ARDUINO_ARCH_ESP32
  #include <Arduino.h>
#else
    #include <driver/gpio.h>
    void digitalWrite(uint8_t gpio_num, uint32_t level)
    {
        gpio_set_level((gpio_num_t) gpio_num, level);
    }
#endif
```

I am just playing around with this library, esp32-S3 and my USB library.

Thanks

mrfaptastic, 2022-12-17:
Absolutely. Please raise a pull request.

chegewara, 2022-12-17:
Here is example app:
https://github.com/chegewara/auxiliary-display

Example is prepared for virtual matrix with 2x2 panels, but can be changed in main.cpp.

closed 2022-12-17 by chegewara