mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #355: Arduino IDE 2.0 does not compile ESP32

Serega88kos, 2022-12-24:
Hello, @mrfaptastic ))
```
\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32/include/driver/include/driver/i2s.h:16:0,
\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/esp32_i2s_parallel_dma.h:20,
\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/ESP32-HUB75-MatrixPanel-I2S-DMA.h:14,
from C:\UserTemp\.arduinoIDE-unsaved20221124-8068-plxvet.owrb\AuroraDemo\AuroraDemo.ino:1:
\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:95:31: error: expected '}' before '__attribute__'
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
                               ^
\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:95:59: error: expected unqualified-id before '=' token
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
                                                           ^
\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:101:3: error: 'i2s_comm_format_t' does not name a type
 } i2s_comm_format_t;
   ^
In file included \libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/esp32_i2s_parallel_dma.h:20:0,
                 \libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/ESP32-HUB75-MatrixPanel-I2S-DMA.h:14,
                 from C:\UserTemp\.arduinoIDE-unsaved20221124-8068-plxvet.owrb\AuroraDemo\AuroraDemo.ino:1:
\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32/include/driver/include/driver/i2s.h:98:5: error: 'i2s_comm_format_t' does not name a type
     i2s_comm_format_t       communication_format;       /*!< I2S communication format */
     ^
In file included \libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/ESP32-HUB75-MatrixPanel-I2S-DMA.h:14:0,
                 from C:\UserTemp\.arduinoIDE-unsaved20221124-8068-plxvet.owrb\AuroraDemo\AuroraDemo.ino:1:
\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/esp32_i2s_parallel_dma.h:102:1: error: expected declaration before '}' token
 }
 ^
cc1plus.exe: warning: unrecognized command line option '-Wno-frame-address'
exit status 1
Compilation error: exit status 1
```
Installed the latest version of the kernel, also does not work.
Tried commenting out deprecated attribute in **i2s_types.h**
The errors are gone, but this is what happened.

```
/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find crt1-sim.o: No such file or directory
/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find _vectors.o: No such file or directory
collect2.exe: error: ld returned 1 exit status
```
This can be fixed in the library, or it is error in Adafruit_GFX?

mrfaptastic, 2022-12-24:
Looks like an error with the aurora demo? Can you try compiling the simple test shapes demo?

Can you also check that you have the latest version of the esp32 board library installed.

Serega88kos, 2022-12-25:
Problem with all sketches. Even my firmware is no longer compiled.
The kernel has tried both the latest 2.0.6 and 2.0.5 and older 2.0.1.
In the old Arduino IDE 1.8.19 version it compiled without problems.

Arduino IDE 2.0.3
```
In file included from \1_SimpleTestShapes.ino.cpp:1:0:
\esp32\2.0.6\cores\esp32/Arduino.h:184:12: error: 'std::round' has not been declared
 using std::round;
            ^
In file included from \libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:11:0,
                 from \1_SimpleTestShapes\1_SimpleTestShapes.ino:6:
\libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/platforms/platform_detect.hpp:58:19: note: #pragma message: Compiling for original ESP32 (2015 release)
   #pragma message "Compiling for original ESP32 (2015 release)"  
                   ^
In file included from \esp32\2.0.6/tools/sdk/esp32/include/driver/include/driver/i2s.h:16:0,
                 from \libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/platforms/esp32/esp32_i2s_parallel_dma.hpp:46,
                 from \libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/platforms/platform_detect.hpp:62,
                 from \libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:11,
                 from \1_SimpleTestShapes\1_SimpleTestShapes.ino:6:
\esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:87:31: error: expected '}' before '__attribute__'
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
                               ^
\esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:87:59: error: expected unqualified-id before '=' token
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
                                                           ^
\esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:93:3: error: 'i2s_comm_format_t' does not name a type
 } i2s_comm_format_t;
   ^
\esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:199:1: error: expected declaration before '}' token
 }
 ^
exit status 1
Compilation error: exit status 1
```

Serega88kos, 2022-12-25:
Installed Arduino 1.8.19 library 3.0.0 the problem is similar.
Installed Arduino 1.8.19 library 2.0.7 the problem is similar.
```
In file included from sketch\1_SimpleTestShapes.ino.cpp:1:0:
esp32\2.0.6\cores\esp32/Arduino.h:184:12: error: 'std::round' has not been declared
 using std::round;
            ^
In file included from esp32\2.0.6/tools/sdk/esp32/include/driver/include/driver/i2s.h:16:0,
                 \libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/esp32_i2s_parallel_dma.h:20,
                 \libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/ESP32-HUB75-MatrixPanel-I2S-DMA.h:14,
                 \libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\examples\1_SimpleTestShapes\1_SimpleTestShapes.ino:6:
esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:87:31: error: expected '}' before '__attribute__'
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
                               ^
esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:87:59: error: expected unqualified-id before '=' token
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
                                                           ^
esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:93:3: error: 'i2s_comm_format_t' does not name a type
 } i2s_comm_format_t;
   ^
In file included from esp32\2.0.6/tools/sdk/esp32/include/driver/include/driver/adc.h:14:0,
                 from esp32\2.0.6/tools/sdk/esp32/include/driver/include/driver/i2s.h:21,
                 \libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/esp32_i2s_parallel_dma.h:20,
                 \libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display/ESP32-HUB75-MatrixPanel-I2S-DMA.h:14,
                 \libraries\ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display\examples\1_SimpleTestShapes\1_SimpleTestShapes.ino:6:
esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/adc_types.h:90:27: error: expected '}' before '__attribute__'
     ADC_DIGI_FORMAT_12BIT __attribute__((deprecated)),  /*!<ADC to DMA data format,                [15:12]-channel, [11: 0]-12 bits ADC data (`adc_digi_output_data_t`). Note: For single convert mode. */
                           ^
esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/adc_types.h:90:54: error: expected unqualified-id before ',' token
     ADC_DIGI_FORMAT_12BIT __attribute__((deprecated)),  /*!<ADC to DMA data format,                [15:12]-channel, [11: 0]-12 bits ADC data (`adc_digi_output_data_t`). Note: For single convert mode. */
                                                      ^
esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/adc_types.h:94:5: warning: 'ADC_DIGI_FORMAT_11BIT' is deprecated [-Wdeprecated-declarations]
     ADC_DIGI_OUTPUT_FORMAT_TYPE1,   ///< See `adc_digi_output_data_t.type1`
     ^
esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/adc_types.h:89:14: note: declared here
 typedef enum {
              ^
esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/adc_types.h:95:5: warning: 'ADC_DIGI_FORMAT_11BIT' is deprecated [-Wdeprecated-declarations]
     ADC_DIGI_OUTPUT_FORMAT_TYPE2,   ///< See `adc_digi_output_data_t.type2`
     ^
esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/adc_types.h:89:14: note: declared here
 typedef enum {
              ^
esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/adc_types.h:96:1: error: expected unqualified-id before '}' token
 } adc_digi_output_format_t;
 ^
esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/adc_types.h:96:1: error: expected declaration before '}' token
cc1plus.exe: warning: unrecognized command line option '-Wno-frame-address'
exit status 1
Ошибка компиляции для платы DOIT ESP32 DEVKIT V1.
```

Serega88kos, 2022-12-25:
Maybe something affects from Adafruit_GFX?

mrfaptastic, 2022-12-25:
No idea what the issue is. I use PlatformIO, but will install Arduino 2.0 in the new year to test.

mrfaptastic, 2023-01-01:
Hi @Serega88kos  - I found the issue and have fixed the library with ce2b6264dd77c15f0f533fe6587b84dd403f8013 

Feel free to confirm as fixed as well.

closed 2023-01-01 by mrfaptastic

Serega88kos, 2023-01-02:
Compiling for ESP32-S2 is successful.
In the coming updates, you can add to other esp32 boards.
Thank you!

mrfaptastic, 2023-01-02:
The issue was specific to the S2. Other boards worked OK.

Serega88kos, 2023-01-03:
DOIT ESP32 DEVKIT V1 does not compile. it is mainly chosen for firmware?

Serega88kos, 2023-01-03:
```
In file included from 1_SimpleTestShapes.ino.cpp:1:0:
\esp32\2.0.6\cores\esp32/Arduino.h:184:12: error: 'std::round' has not been declared
 using std::round;
            ^
In file included from \src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:12:0,
                 from \1_SimpleTestShapes\1_SimpleTestShapes.ino:6:
\src/platforms/platform_detect.hpp:58:19: note: #pragma message: Compiling for original ESP32 (released 2016)
   #pragma message "Compiling for original ESP32 (released 2016)"  
                   ^
In file included from \esp32\2.0.6/tools/sdk/esp32/include/driver/include/driver/i2s.h:16:0,
                 from \src/platforms/esp32/esp32_i2s_parallel_dma.hpp:46,
                 from \src/platforms/platform_detect.hpp:62,
                 from \src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:12,
                 from \1_SimpleTestShapes\1_SimpleTestShapes.ino:6:
\esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:87:31: error: expected '}' before '__attribute__'
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
                               ^
\esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:87:59: error: expected unqualified-id before '=' token
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
                                                           ^
\esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:93:3: error: 'i2s_comm_format_t' does not name a type
 } i2s_comm_format_t;
   ^
\esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:199:1: error: expected declaration before '}' token
 }
 ^
exit status 1
Compilation error: exit status 1
```

mrfaptastic, 2023-01-03:
Interesting. Are you using Arduino IDE (Version 2) with latest ESP 32 platform libraries?

If not, please confirm what IDE and environment.

Serega88kos, 2023-01-03:
Arduino IDE 2.0.3
ESP32 Core 2.0.6

mrfaptastic, 2023-01-03:
Just tried and I do not get an error when compiling with the same configuration as you have provided with version 3.0.2 of this library.

mrfaptastic, 2023-01-03:
3.0.2 was released yesterday which did fix a bunch of issues.

Serega88kos, 2023-01-24:
```
In file included from \1_SimpleTestShapes.ino.cpp:1:0:
\esp32\2.0.6\cores\esp32/Arduino.h:184:12: error: 'std::round' has not been declared
 using std::round;
            ^
In file included from \src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:12:0,
                 from \1_SimpleTestShapes.ino:6:
\src/platforms/platform_detect.hpp:58:19: note: #pragma message: Compiling for original ESP32 (released 2016)
   #pragma message "Compiling for original ESP32 (released 2016)"  
                   ^
In file included from \esp32\2.0.6/tools/sdk/esp32/include/driver/include/driver/i2s.h:16:0,
                 from \src/platforms/esp32/esp32_i2s_parallel_dma.hpp:46,
                 from \src/platforms/platform_detect.hpp:62,
                 from \src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:12,
                 from \1_SimpleTestShapes.ino:6:
\esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:87:31: error: expected '}' before '__attribute__'
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
                               ^
\esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:87:59: error: expected unqualified-id before '=' token
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
                                                           ^
\esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:93:3: error: 'i2s_comm_format_t' does not name a type
 } i2s_comm_format_t;
   ^
\esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:199:1: error: expected declaration before '}' token
 }
 ^
exit status 1
Compilation error: exit status 1
```
DOIT ESP32 DEVKIT V1
3.0.5

solhuebner, 2023-01-24:
Seems like an issue that is coming up from time to time: https://github.com/espressif/arduino-esp32/issues/7077

Andrew170285, 2023-03-30:
I have panels of 80*40 pixels, 1/20 scan. where can I set up or fix the "ABCDE" account?

mrfaptastic, 2023-04-17:
Sorry @Serega88kos, I don't understand.

Serega88kos, 2023-04-17:
With this code compilation in esp32 is successful.

Serega88kos, 2023-04-17:
ESP32-HUB75-MatrixPanel-I2S-DMA.cpp

mrfaptastic, 2023-04-17:
Feel free to comment that code out if it causes issues for you. I don't have that issue myself but will take a look some time.