mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #359: Unable to compile for ESP-S2

russdx, 2022-12-28:
Hello

I have a [ESP32 S2 Mini V1.0.0 Board](https://kunkune.co.uk/shop/esp32-esp8266/esp32-s2-mini-v1-0-0-board) and read this library supports the ESP-S2/S3
I am using Arduino 1.8.10, with ESP 2.0.6 Board libs installed and the latest version of this lib.

I am able to compile test examples using a standard ESP32 board, but when I select any type of ESPS2/S3 boards it throws errors with lots of missing defines (sort of like it does not support these chips...) 

It does not appear to like the Bus_Parallel16?

Any ideas where I am going wrong would be much appreciated.

Regards
Russell

Arduino ERROR
`Arduino: 1.8.10 (Windows 10), Board: "ESP32S2 Dev Module, FTDI Adapter, Disabled, Disabled, Disabled, UART0, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None, Disabled"

In file included from C:\Users\russd\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:12,
                 from C:\Users\russd\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\examples\1_SimpleTestShapes\1_SimpleTestShapes.ino:6:
C:\Users\russd\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src/platforms/platform_detect.hpp:31:19: note: #pragma message: Compiling for ESP32-S2
   #pragma message "Compiling for ESP32-S2"
                   ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\russd\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src\ESP32-HUB75-MatrixPanel-I2S-DMA.h:12,
                 from C:\Users\russd\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src\ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:1:
C:\Users\russd\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src\platforms/platform_detect.hpp:31:19: note: #pragma message: Compiling for ESP32-S2
   #pragma message "Compiling for ESP32-S2"
                   ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\russd\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src\ESP32-HUB75-MatrixPanel-I2S-DMA.h:12,
                 from C:\Users\russd\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src\ESP32-HUB75-MatrixPanel-leddrivers.cpp:13:
C:\Users\russd\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src\platforms/platform_detect.hpp:31:19: note: #pragma message: Compiling for ESP32-S2
   #pragma message "Compiling for ESP32-S2"
                   ^~~~~~~~~~~~~~~~~~~~~~~~
c:/users/russd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-HUB75-MatrixPanel-DMA-master\ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o:(.literal._ZN19MatrixPanel_I2S_DMA12configureDMAERK13HUB75_I2S_CFG+0xc): undefined reference to `Bus_Parallel16::create_dma_desc_link(void*, unsigned int, bool)'

c:/users/russd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-HUB75-MatrixPanel-DMA-master\ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o:(.literal._ZN19MatrixPanel_I2S_DMA12configureDMAERK13HUB75_I2S_CFG+0x10): undefined reference to `Bus_Parallel16::config(Bus_Parallel16::config_t const&)'

c:/users/russd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-HUB75-MatrixPanel-DMA-master\ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o:(.literal._ZN19MatrixPanel_I2S_DMA12configureDMAERK13HUB75_I2S_CFG+0x14): undefined reference to `Bus_Parallel16::init()'

c:/users/russd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-HUB75-MatrixPanel-DMA-master\ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o:(.literal._ZN19MatrixPanel_I2S_DMA12configureDMAERK13HUB75_I2S_CFG+0x18): undefined reference to `Bus_Parallel16::dma_transfer_start()'

c:/users/russd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-HUB75-MatrixPanel-DMA-master\ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o:(.literal._ZN19MatrixPanel_I2S_DMA12configureDMAERK13HUB75_I2S_CFG+0x1c): undefined reference to `Bus_Parallel16::flip_dma_output_buffer(int&)'

c:/users/russd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-HUB75-MatrixPanel-DMA-master\ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o:(.literal._ZN19MatrixPanel_I2S_DMA17allocateDMAmemoryEv+0x20): undefined reference to `Bus_Parallel16::enable_double_dma_desc()'

c:/users/russd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-HUB75-MatrixPanel-DMA-master\ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o:(.literal._ZN19MatrixPanel_I2S_DMA17allocateDMAmemoryEv+0x24): undefined reference to `Bus_Parallel16::allocate_dma_desc_memory(unsigned int)'

c:/users/russd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-HUB75-MatrixPanel-DMA-master\ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o: in function `MatrixPanel_I2S_DMA::configureDMA(HUB75_I2S_CFG const&)':

C:\Users\russd\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:183: undefined reference to `Bus_Parallel16::create_dma_desc_link(void*, unsigned int, bool)'

c:/users/russd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\russd\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:187: undefined reference to `Bus_Parallel16::create_dma_desc_link(void*, unsigned int, bool)'

c:/users/russd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-HUB75-MatrixPanel-DMA-master\ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o: in function `rowBitStruct::size(unsigned char)':

C:\Users\russd\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:134: undefined reference to `Bus_Parallel16::create_dma_desc_link(void*, unsigned int, bool)'

c:/users/russd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\russd\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:134: undefined reference to `Bus_Parallel16::create_dma_desc_link(void*, unsigned int, bool)'

c:/users/russd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\russd\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:134: undefined reference to `Bus_Parallel16::create_dma_desc_link(void*, unsigned int, bool)'

c:/users/russd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-HUB75-MatrixPanel-DMA-master\ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o:C:\Users\russd\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:201: more undefined references to `Bus_Parallel16::create_dma_desc_link(void*, unsigned int, bool)' follow

c:/users/russd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: libraries\ESP32-HUB75-MatrixPanel-DMA-master\ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o: in fun [... cut ...]
c:/users/russd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: sketch\1_SimpleTestShapes.ino.cpp.o: in function `MatrixPanel_I2S_DMA::~MatrixPanel_I2S_DMA()':

C:\Users\russd\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:371: undefined reference to `Bus_Parallel16::release()'

collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32S2 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

mrfaptastic, 2022-12-28:
I'm on the road until the new year so can't test, but can you stick the following line in line 2 of the ESP32-HUB75-MatrixPanel-I2S-DMA.cpp and see if that helps?

```
#include "platforms/platform_detect.hpp"
```

russdx, 2022-12-28:
No worries thanks for the quick reply

I can see it is switching as I get the output '#pragma message: Compiling for ESP32-S2' So definitely switching to the S2 but any method under 'Bus_Parallel16' appears to then be missing.

mrfaptastic, 2022-12-28:
Yeah I'm not sure what is going on there. The linker isn't happy for whatever reason. Works in PlatformIO which is what I use namely.

russdx, 2022-12-28:
Ah i'll give it a go on PlatformIO :)

mrfaptastic, 2023-01-01:
Hi @russdx - I found the issue and have fixed the library with ce2b6264dd77c15f0f533fe6587b84dd403f8013 

Feel free to confirm as fixed as well.

closed 2023-01-01 by mrfaptastic

russdx, 2023-01-01:
> Hi @russdx - I found the issue and have fixed the library with [ce2b626](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/commit/ce2b6264dd77c15f0f533fe6587b84dd403f8013)
> 
> Feel free to confirm as fixed as well.

Now compiles perfectly under Arduino IDE
Thank you for fixing this.

mrfaptastic, 2023-01-01:
Not a problem. Happy New Year!

utiq, 2024-01-08:
@mrfaptastic but now it doesn't work in Platformio

```
ESP32-HUB75-MatrixPanel-DMA/src/platforms/platform_detect.hpp:38:19: note: #pragma message: Compiling for ESP32-S3
   #pragma message "Compiling for ESP32-S3"
                   ^~~~~~~~~~~~~~~~~~~~~~~~
Linking .pio/build/my-project/firmware.elf
/Users/me/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/my-project/libf32/libHUB75.a(ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o):(.literal._ZN19MatrixPanel_I2S_DMA12configureDMAERK13HUB75_I2S_CFG+0xc): undefined reference to `Bus_Parallel16::create_dma_desc_link(void*, unsigned int, bool)'
/Users/me/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/my-project/libf32/libHUB75.a(ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o):(.literal._ZN19MatrixPanel_I2S_DMA12configureDMAERK13HUB75_I2S_CFG+0x10): undefined reference to `Bus_Parallel16::config(Bus_Parallel16::config_t const&)'
/Users/me/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/my-project/libf32/libHUB75.a(ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o):(.literal._ZN19MatrixPanel_I2S_DMA12configureDMAERK13HUB75_I2S_CFG+0x14): undefined reference to `Bus_Parallel16::init()'
/Users/me/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/my-project/libf32/libHUB75.a(ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o):(.literal._ZN19MatrixPanel_I2S_DMA12configureDMAERK13HUB75_I2S_CFG+0x18): undefined reference to `Bus_Parallel16::dma_transfer_start()'
/Users/me/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/my-project/libf32/libHUB75.a(ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o):(.literal._ZN19MatrixPanel_I2S_DMA17allocateDMAmemoryEv+0x50): undefined reference to `Bus_Parallel16::enable_double_dma_desc()'
/Users/me/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pio/build/my-project/libf32/libHUB75.a(ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o):(.literal._ZN19MatrixPanel_I2S_DMA17allocateDMAmemoryEv+0x54): undefined reference to `Bus_Parallel16::allocate_dma_desc_memory(unsigned int)'
.
.
.
```