mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #362: Wrong colors

fruktiz32, 2023-01-03:
Hello, I have been using many P2.5 panels without any problems for long time. Now I got P5 panel and it was working fine with same software as P2.5 without any issues. Now I updated ESP32 library and MATRIX library and some color does not work on P5 panel, when P2.5 works fine.  I have tried to upload SimpleShapes example to P5 and all colors works, but with my software it does work only on P2.5 but not P5. I'm so confused, any ideas?

mrfaptastic, 2023-01-03:
Need more details.

* What ESP device your using
* Library version

Have you checked the pin mapping?

fruktiz32, 2023-01-03:
> 

I have checked many times wiring. But that is weird that example method works totally fine with same wiring, then I upload my sketch and colors are wrong..

1) ESP32 is 30 pin DEVKITV1 (have tried another esp, but result the same).
2) I have tried same device on P2.5 and P5, but P5 colors are wrong.
3) Matrix library version is 2.0.7 (do not use newest because it goes to restart loop with 3.x.x versions)

I have checked mapping, it is the same as mentioned in wiring tutorial.

mrfaptastic, 2023-01-03:
Can you please provide detailed debug output when using version 3.0.2? I wish to understand the crash that is occurring on boot.

  I can't comment on what is causing the issue when using 2.0.7 as I don't know what version you were using before that.

Also, there are plenty of HUB75 boards floating around where the colour pins are the wrong way around - not much can be done than changing the pin mapping when using P5.

fruktiz32, 2023-01-03:
Haha. Now I understand what happened. I always using P2.5 and it was faulty colors too and I switched green to blue and blue to green. And was wondering why my P5 panel shows wrong colors.. :D Now I understand that P2.5 is faulty, not P5. Thank you for your time..

closed 2023-01-03 by fruktiz32

mrfaptastic, 2023-01-03:
Why chance you can send me the debug output of this crash you get with version 3.0.x of the library?

I've taken the time to help you, so would you be so kind to enlighten me on what the issue with 3x version?

There could be some hidden bug I don't know about.

fruktiz32, 2023-01-03:
> 

[image]

I attatch my wiring scheme. With 2.0.7 I can run with external MicroUSB, but when I updated it to 3.0.x does not work anymore with external USB anymore, it works only with ESP32 port.

mrfaptastic, 2023-01-03:
Ahhhhhhhhhh.... Yes. I've run into this as well.

I think we increased the default brightness a bit with 3.x which must trigger this issue.

Use set brightness function before displaying anything (set it low) to see if this helps. Then increase brightness.

Also, it helps to have a capacitor across VCC and gnd to smooth power draw spikes these panels cause.