mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #363: Inverted Blue/Green colors

GrimorJander, 2023-01-06:
Hi everybody and happy 2023!
I just switched from a P4 panel to a P2.5 with ICN2047 driver.
With the standard wiring of this library all previous projects are working OK, and the only issue is when I draw pure blue color, it shows green, and viceversa (green is blue now). So, all mixed colors seen way off, like inverted.
¿What do you think will be the easier fix? Changing B1 and B2 for G1 and G2 on the code? Or rewiring and finding the wrong pin? Thanks in advance

GrimorJander, 2023-01-06:
Well, i find that rewiring B1 and B2 as G1 and G2 solved the problem. (B1=26, G1=27, B2=12, G2=13)
But somehow i can´t make  those changes via software, on "gif animation" example.

GrimorJander, 2023-01-07:
Now I would like to rephrase my question:
Why can't I make the remapping work from the code?

`// Example sketch which shows how to display a 64x32 animated GIF image stored in FLASH memory
// on a 64x32 LED matrix
//
// Credits: https://github.com/bitbank2/AnimatedGIF/tree/master/examples/ESP32_LEDMatrix_I2S
// 

/* INSTRUCTIONS
 *
 * 1. First Run the 'ESP32 Sketch Data Upload Tool' in Arduino from the 'Tools' Menu.
 *    - If you don't know what this is or see it as an option, then read this:
 *      https://github.com/me-no-dev/arduino-esp32fs-plugin 
 *    - This tool will upload the contents of the data/ directory in the sketch folder onto
 *      the ESP32 itself.
 *
 * 2. You can drop any animated GIF you want in there, but keep it to the resolution of the 
 *    MATRIX you're displaying to. To resize a gif, use this online website: https://ezgif.com/
 *
 * 3. Have fun.
 */

#define FILESYSTEM SPIFFS
#include <SPIFFS.h>
#include <AnimatedGIF.h>
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

// ----------------------------

/* 
 * Below is an is the 'legacy' way of initialising the MatrixPanel_I2S_DMA class.
 * i.e. MATRIX_WIDTH and MATRIX_HEIGHT are modified by compile-time directives.
 * By default the library assumes a single 64x32 pixel panel is connected.
 *
 * Refer to the example '2_PatternPlasma' on the new / correct way to setup this library
 * for different resolutions / panel chain lengths within the sketch 'setup()'.
 * 
 */

// Change these to whatever suits
#define R1_PIN 25
#define G1_PIN 27
#define B1_PIN 26
#define R2_PIN 14
#define G2_PIN 13
#define B2_PIN 12
#define A_PIN 23
#define B_PIN 19 // Changed from library default
#define C_PIN 5
#define D_PIN 17
#define E_PIN -1
#define LAT_PIN 4
#define OE_PIN 15
#define CLK_PIN 16
  // Panel configuration
  #define PANEL_RES_X 64      // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32     // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1 
  
  
  #define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
  #define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW
  
  //  NOTE: DEFAULT EXAMPLE SETUP IS FOR A CHAIN OF TWO x 1/8 SCAN PANELS
    
  // Change this to your needs, for details on VirtualPanel pls read the PDF!
  #define SERPENT true
  #define TOPDOWN false
  
[... 236 lines cut ...]
            while (gifFile)
            {
              if (!gifFile.isDirectory()) // play it
              {
                
                // C-strings... urghh...                
                memset(filePath, 0x0, sizeof(filePath));                
                strcpy(filePath, gifFile.path());
                
                // Show it.
                ShowGIF(filePath);
               
              }
              gifFile.close();
              gifFile = root.openNextFile();
            }
         root.close();
      } // root
      
      delay(1000); // pause before restarting
      
   } // while
}`

I think i´m missing something because changing pin numbers make no effect .

mrfaptastic, 2023-01-07:
Can't see where your assign the '_pins' variable you declare, to the mxconfig struct.

GrimorJander, 2023-01-07:
> Can't see where your assign the value of the '_pins' variable you declare, to the mxconfig struct.

Sure it will be the failure. 
Is this on the begining of the "void setup"?
I´m very noob, so, do you have an example where custom GPIOs are declared this way like you said?
Thanks again @mrfaptastic

mrfaptastic, 2023-01-07:
Take a look at the bitmapicons example. It has (at least commented out) where you pass the _pins variable

GrimorJander, 2023-01-07:
Well, i see.
I´ve added :
`
HUB75_I2S_CFG::i2s_pins _pins={R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN};
HUB75_I2S_CFG mxconfig(
           PANEL_RES_X,   // module width
           PANEL_RES_Y,   // module height
           PANEL_CHAIN,    // Chain length
           _pins,   // pin mapping
  HUB75_I2S_CFG::FM6126A      // driver chip
 );
` 
And still no PIN has changed

mrfaptastic, 2023-01-07:
It's really clear where you have the mxconfig from your code dump, you only need on within the setup()

```
 HUB75_I2S_CFG::i2s_pins _pins = {
  R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, 
  A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, 
  LAT_PIN, OE_PIN, CLK_PIN
 };

HUB75_I2S_CFG mxconfig
(
PANEL_RES_X, // module width
PANEL_RES_Y, // module height
PANEL_CHAIN, // Chain length
_pins
);

mxconfig.clkphase = false;
mxconfig.driver = HUB75_I2S_CFG::ICN2038S;

```

GrimorJander, 2023-01-07:
Oh yeah now it's clear and working as it should.
Excuse me for my incompetence.
Now I see that other issues I have opened are also due to my lack of programming expertise:
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/issues/302
I leave the code of the Gif animation example with the custom pins (green and blue channels swapped) in case it is useful to someone in the future:

`// Example sketch which shows how to display a 64x32 animated GIF image stored in FLASH memory
// on a 64x32 LED matrix
//
// Credits: https://github.com/bitbank2/AnimatedGIF/tree/master/examples/ESP32_LEDMatrix_I2S

#define FILESYSTEM SPIFFS
#include <SPIFFS.h>
#include <AnimatedGIF.h>
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

// Change these to whatever suits
#define R1_PIN 25
#define G1_PIN 27
#define B1_PIN 26
#define R2_PIN 14
#define G2_PIN 13
#define B2_PIN 12
#define A_PIN 23
#define B_PIN 19 // Changed from library default
#define C_PIN 5
#define D_PIN 17
#define E_PIN -1
#define LAT_PIN 4
#define OE_PIN 15
#define CLK_PIN 16
  // Panel configuration
  #define PANEL_RES_X 64      // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32     // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1 
  
  
  #define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
  #define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW
  
  
    
  // Change this to your needs, for details on VirtualPanel pls read the PDF!
  #define SERPENT true
  #define TOPDOWN false
  
  // placeholder for the matrix object
  MatrixPanel_I2S_DMA *dma_display = nullptr;
 

uint16_t myBLACK = dma_display->color565(0, 0, 0);
uint16_t myWHITE = dma_display->color565(255, 255, 255);
uint16_t myRED = dma_display->color565(255, 0, 0);
uint16_t myGREEN = dma_display->color565(0, 255, 0);
uint16_t myBLUE = dma_display->color565(0, 0, 255);

AnimatedGIF gif;
File f;
int x_offset, y_offset;

// Draw a line of image directly on the LED Matrix
void GIFDraw(GIFDRAW *pDraw)
{
    uint8_t *s;
    uint16_t *d, *usPalette, usTemp[320];
    int x, y, iWidth;

  iWidth = pDraw->iWidth;
  if (iWidth > MATRIX_WIDTH)
      iWidth = MATRIX_WIDTH;
[... 206 lines cut ...]
            {
              if (!gifFile.isDirectory()) // play it
              {
                
                // C-strings... urghh...                
                memset(filePath, 0x0, sizeof(filePath));                
                strcpy(filePath, gifFile.path());
                
                // Show it.
                ShowGIF(filePath);
               
              }
              gifFile.close();
              
              gifFile = root.openNextFile();
            }
         root.close();
      } // root
      
      delay(1000); // pause before restarting
      
   } // while
}`

Thank you very much once again, and the thread can now be closed.

closed 2023-01-07 by GrimorJander