mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #366: Cannot set panel brightness using the NO_GFX option.

BoltSwith, 2023-01-07:
Hi,

Great Library here and many thanks for sharing.

Using a 128x64 panel with the ICN2037BP driver chip, I was able to successfully compile the project and try some of the examples supplied with the library (fillscreens, bouncing_squares, pattern_plasma, Simple_Test_Shapes). All examples worked fine, however, attempting to set panel brightness using both "setBrightness8();" and "setPanelBrightness()" with Adafruit_GFX_Library and NO_GFX option seem not to have any effect on actually panel brightness (brightness remains the same at both 1 and 255 values). 

Trying the patten_Plasma example using the FastLED and GFX_Root Libs, I was able to adjust panel brightness conveniently.

Does the "setBrightness" functions of this library depend on other APIs - Adafruit_GFX_Library, FastLED or Neomatrix - to adjust actual panel brightness? 
I would prefer to use the NO_GFX option and implement my own graphics/text drawing functions and the native methods - setBrightness(), drawPixel(), clearScreen(), fillScreen() - provided by this library. Thanks.

mrfaptastic, 2023-01-07:
Hi @BoltSwith 

That's really bizarre.... the setPanelBrightness and setBrightness8 are defined within [the main header file](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/blob/master/src/ESP32-HUB75-MatrixPanel-I2S-DMA.h) and should be available regardless of the compile option, and have no dependency on any GFX library.

Not sure why it doesn't work Adafruit_GFX_Library alone. I was testing dimming earlier today for a similar issue that was raised, but used Pattern Plasma as my test sketch, and it worked which you corroborate.... but sounds like it doesn't when FastLED isn't available. 

Can you stick a couple of Serial.println's into the brightnesss functions and confirm they're being called properly etc? No idea what would cause this issue.

mrfaptastic, 2023-01-07:
Try the latest git version of the library and see what happens.

BoltSwith, 2023-01-08:
Hurray.....!!!!

Trying the latest git version as you advised, I was able to seamlessly set the brightness of the panel using the "setBrightness()" method. Values tested include 10, 100, 150 and 200. This test was done using the NO_GFX option. 

Many thanks for the swift response.

[image]

BoltSwith, 2023-01-08:
I think an update to this instruction on setting Brightness might be in order.

[Panel Brightness Instruction/Guide](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA#panel-brightness)

closed 2023-01-08 by BoltSwith