mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #370: BluetoothA2DPSink.h not working with  ledcSetup() / ledcWriteNote() ?...

Knottis, 2023-01-11:
### When I initialize the following function, then Bluetooth doesn't work and the CPU crash.

- ledcSetup(); 
- ledcWriteNote()

I hope you can help me to figerout why these are not compatible with each other!..

ledcAttachPin(AlarmPin, ledChannel);

**I use the following equipment:**

- Arduino IDE 2.0.3
- ESP32-WROOM-32D on the ESP32 Dev Module

**Thanks in advance**
Regards, Knottis
_(Sorry for my bad English)_

My program
=========================================================================

#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;    // DAC-1 DAC-2 -> pin-25, Pin-26  "Audio Sound Speaker" on this pin.

const int AlarmPin = 33;      // The PWM pin-33 ->  My "Alarm Speaker" is conected to this pin.

const int freq = 10000;     // set the frequency for 5kHz
const int ledChannel = 0;   // set the PWM channel
const int resolution = 8;   // set PWM resolution

char BLTNAME[60] = "Audio-Speaker"; 

void setup(){
  // Configure LED PWM functionalitites
  // Define the PWM Setup
  ledcSetup(ledChannel, freq, resolution);  // define the PWM Setup

  // Bluetooth config
  const i2s_config_t i2s_config = {
      .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN),
      .sample_rate = 44100,                           // corrected by info from bluetooth
      .bits_per_sample = (i2s_bits_per_sample_t) 16,  // the DAC module will only take the 8bits from MSB
      .channel_format =  I2S_CHANNEL_FMT_RIGHT_LEFT,
      .communication_format = (i2s_comm_format_t) I2S_COMM_FORMAT_STAND_MSB,
      .intr_alloc_flags = 0,                          // default interrupt priority
      .dma_buf_count = 8,
      .dma_buf_len = 64,
      .use_apll = false,
      //.use_apll = true,
      .tx_desc_auto_clear = true // avoiding noise in case of data unavailability
  };

  a2dp_sink.set_i2s_config(i2s_config);
  
  a2dp_sink.set_mono_downmix(true);
  
  // START UP BLUETOOTH DEVICES.....
  a2dp_sink.start(BLTNAME, false);        // true or false

  // Changes the volume -> 100%
  a2dp_sink.set_volume(100);              // Changes the volume (use the range 0-100) 0...100%
}
 

void loop(){
    
    ledcAttachPin(AlarmPin, ledChannel);
    //
    ledcWriteNote(ledChannel, NOTE_C, 7);
    delay(500);
    ledcWriteNote(ledChannel, NOTE_Cs, 7);
    delay(500);
    ledcWriteNote(ledChannel, NOTE_E, 7);
    delay(500);
    //
    ledcDetachPin(AlarmPin);

    delay(5000);
}

ERROR
=========================================================================

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
E (371) esp_core_dump_fl���B� Core dump data check failed:
Calculated checksum='ce03c375'
Image checksum='ffffffff'
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.

Core  1 register dump:
PC      : 0x400dbfdf  PS      : 0x00060034  A0      : 0x800dc6f1  A1      : 0x3ffbd3c0  
A2      : 0x40084874  A3      : 0x001f8000  A4      : 0x000007d0  A5      : 0x601f8000  
A6      : 0x3ffbd6ac  A7      : 0x00000000  A8      : 0x40084873  A9      : 0x3ffbd380  
A10     : 0x00000000  A11     : 0x3ffbd3ac  A12     : 0x0000000a  A13     : 0x3ffbd3b8  
A14     : 0x3f4035e9  A15     : 0x00000000  SAR     : 0x0000001e  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40084873  LBEG    : 0x40090494  LEND    : 0x4009049f  LCOUNT  : 0xffffffff  

Backtrace: 0x400dbfdc:0x3ffbd3c0 0x400dc6ee:0x3ffbd3e0 0x400dc117:0x3ffbd400 0x400dc543:0x3ffbd470 0x40083965:0x3ffbd4c0 0x4008cc69:0x3ffbd4e0 0x00040021:0x3ffbd5a0 |<-CORRUPTED

Core  0 register dump:
PC      : 0x23680569  PS      : 0x33781579  A0      : 0xc3322569  A1      : 0x52357910  
A2      : 0x643710c5  A3      : 0x7803680b  A4      : 0x69338b13  A5      : 0x8b157905  
A6      : 0x07e42755  A7      : 0x0714e417  A8      : 0xf01d21e4  A9      : 0x334b0368  
A10     : 0x554b0569  A11     : 0x0704e417  A12     : 0xf01d11e4  A13     : 0x2b001362  
A14     : 0x00556233  A15     : 0xe407552b  SAR     : 0x00f01d02  EXCCAUSE: 0x62000362  
EXCVADDR: 0xf01d0045  LBEG    : 0x00ffa416  LEND    : 0xbe804023  LCOUNT  : 0xc033b015  

Backtrace: 0x23680566:0x52357910 |<-CORRUPTED

mrfaptastic, 2023-01-11:
You're out of memory most likely.
Bluetooth stack consumes a huge amount of memory. I doubt you'll be able to use that with a LED display at the same time.

Perhaps buy an ESP32-S3 and use the PSRAM option for the LED DMA Buffer.

solhuebner, 2023-01-11:
You could also have a look at https://github.com/bluekitchen/btstack as it should be less memory hungry

closed 2023-01-11 by Knottis