mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #373: Some a bit incorrect color and shade gradations

Kouzeru, 2023-01-19:
I'm putting some sort of image test to display some hue and value spectrum,
I got some noticeable color banding when using this driver:
[image]

I had deployed another technique to actually do the BCM properly, thus, displaying more faithful hue and value spectrum:
[image]
(with the cost of sacrificing 1/4 of the total brightness of course, due to limitation of my driver)

Here;s the code of the test image
```
  char t = millis()>>4;
  for(int x=0;x<128;x++)
    for(int y=0;y<64;y++)
      dma_display->drawPixelRGB888(x,y,
      (uint8_t)((max(min(cosf(2.f*PI*((x-t)/128.f+0.f/3.f))+0.5f,1.f),0.f)*((y+1.f)/64.f))*255.f),
      (uint8_t)((max(min(cosf(2.f*PI*((x-t)/128.f+1.f/3.f))+0.5f,1.f),0.f)*((y+1.f)/64.f))*255.f),
      (uint8_t)((max(min(cosf(2.f*PI*((x-t)/128.f+2.f/3.f))+0.5f,1.f),0.f)*((y+1.f)/64.f))*255.f)
      );
  ```

mrfaptastic, 2023-01-19:
Can you confirm if you get this issue when using version 3.0.0 of the library?

I may have introduced a bug when I fiddled with the way brightness is managed in the most recent version

Kouzeru, 2023-01-19:
I had tested the version 3.0.0; it still yields the exact same result. (I can't take the picture I'm sorry).

But I'd think I could try to reproduce how your driver translate the colors:
[image]

which was plotted pixel by pixel in python code
```py
r = int((max(min(sin((u/8+0/3)*pi*2)+0.5,1),0)*(1-v)/2)*255)
g = int((max(min(sin((u/8+1/3)*pi*2)+0.5,1),0)*(1-v)/2)*255)
b = int((max(min(sin((u/8+2/3)*pi*2)+0.5,1),0)*(1-v)/2)*255)
r0,r1,r2,r3,r4,r5=r>>7&1,r>>6&1,r>>5&1,r>>4&1,r>>3&1,r>>2&1
g0,g1,g2,g3,g4,g5=g>>7&1,g>>6&1,g>>5&1,g>>4&1,g>>3&1,g>>2&1
b0,b1,b2,b3,b4,b5=b>>7&1,b>>6&1,b>>5&1,b>>4&1,b>>3&1,b>>2&1
r,g,b=r0*6+r1*5+r2*4+r3*3+r4*2+r5,g0*6+g1*5+g2*4+g3*3+g4*2+g5,b0*6+b1*5+b2*4+b3*3+b4*2+b5
sqrt(r/21)*256,sqrt(g/21)*256,sqrt(b/21)*256

Kouzeru, 2023-01-20:
The pull request #374 will fix the following artifacts
[image]
[image]
into
[image]
[image]

And I already had thorough tests, and only this is was complaint of the driver.

PR #374 (2023-01-20): BCM accumulation correction
resolves #373 

before correction:
[image]

after correction:
[image]
closed 2023-01-20 by mrfaptastic