mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #377: Clock speed

techwide, 2023-01-22:
Hi,
I am having some weird problem with LED panels using the latest version, 3.0.3. All works fine with 2.0.7.
Some display shows garbage and some display shows some dots with different colors.

[image]

I have tested many different ways and finally I've found that the clock speeds at CLK pin are different. The clock speed of 3.0.3 version is 33MHz but 2.0.7 version is 10MHz. I assume 10MHz is the correct one.

[image]

I've even inserted the following code but the clock is 33MHz.

mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_10M; 

I am not sure whether the clock is adjustable but the clock speed seems to be the problem I am having.

Regards,

Peter

mrfaptastic, 2023-01-22:
Clock speed is adjustable so should be 10mhz.

What esp32 variant are you using?

mrfaptastic, 2023-01-22:
Can you try the latest git version of the library, and check these values versus what you see in the scope?

```
mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_10M;

mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_20M;
```

techwide, 2023-01-22:
Yes the clock now works with HZ_10M and HZ_20M.

BTW..PlatformIO does not detect the latest version. It detects only v3.0.2.

mrfaptastic, 2023-01-23:
Just to be sure HZ_10M gives a 10Mhz clock right?

mrfaptastic, 2023-01-23:
Thanks again for making this library better.

techwide, 2023-01-24:
Yes HZ_10M gives a 10Mhz clock and HZ_20M gives a 20Mhz clock as expected!

closed 2023-01-24 by mrfaptastic