mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #379: HUB12 Support

BimoSora2, 2023-01-28:
Does it support HUB12?

DarrylStrong, 2023-01-28:
Nope, HUB 12 in not multiplexed so totally different.

board707, 2023-01-28:
Not multiplexed? What do you mean?

As far I know, HUB12 use miltplexion with pins A & B. The general algorithm of HUB12 processing is quite similar to work with RGB 32x16 scan4 panels. 
I am work with Mono HUB12 and RGB HUB75 in the same library,

DarrylStrong, 2023-01-28:
Sorry, my mistake.

It looks like a 1/4 scan on second looking.

You will be able to use the library with a few tricks. There is an example of this in the git repository.

Kouzeru, 2023-01-29:
You won't be able to get this work with this library, correctly, anyway.
First of all, the memory layout are different.

while display layout of HUB-75 like this
```
ABCDEF...
ABCDEF...
```

HUB-12 is like this
```
AEIMQU...
BFJNRV...
CGKOSW...
DHLPTX...
```

Also, the polarity of data is inverted, OE is also inverted;
It's way easier for you to work with SPI for HUB-12, and it's easy to make driver for

board707, 2023-01-29:
> layout of HUB-75 like this
> 
> in 1 -> ABCDEF...
> in 2 -> ABCDEF...
> ```

It is true for plain scan RGB panels only, for example for panels where scan is half of height, like 32x16 scan8. If we will talk about 32x16 scan4 panels - many of them has layout similar to monochrome ones.
If you plan to work with wide set of RGB panels - you will forced to manage several variants of memory layout.

Change polarity settings is not difficult at all :)