mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #381: Misleading RAM info in README

LostInCompilation, 2023-01-28:
In the README under supported chips there is this sentence:
```
Original ESP32 - That being the ESP-WROOM-32 module with ESP32‑D0WDQ6 chip from 2017. This MCU has 520kB of SRAM which is much more than all the recent 'reboots' of the ESP32 such as the S2, S3 etc.
```

I think this is a bit misleading. The ESP32-S3 has 512kB of SRAM, instead of 520kB. Maybe the wording should be a bit different. The S2 also has 320kB which is less, but i wouldn't say "much less".

Kouzeru, 2023-01-29:
The sentence is already correct.
ESP32 (this mcu) > ESP32-S3 (one of recent reboots)
ESP32 (this mcu) > (much more than) ESP32-S2

I think it should be elaborated more clearly.

LostInCompilation, 2023-01-29:
The sentence is this in short:
`ESP32‑D0WDQ6 has much more SRAM, than reboots such as S2 and S3.
It's true for S2, but S3 has almost the same amount of RAM as ESP32‑D0WDQ6.

mrfaptastic, 2023-01-30:
Feel free to raise a pull request to update the documentation to wording you think is more appropriate.

LostInCompilation, 2023-01-30:
Sure will think of something tomorrow. Maybe a table with RAM on supported chips and a full list of unsupported chips (RISC family).

mrfaptastic, 2023-02-01:
For now, I've removed this text completely.

closed 2023-02-01 by mrfaptastic

LostInCompilation, 2023-02-02:
I think the new wording is fine. Also the supported chips are clear. Should I still add a table of supported chips and their corresponding RAM or do you think it is fine like it is now?

mrfaptastic, 2023-02-02:
Totally up to you! I'm cool either way.