mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #383: Weird blank horizontal strips for every 8th rows; on deeper color bits

Kouzeru, 2023-01-29:
https://user-images.githubusercontent.com/46141631/215348041-1ae35171-f508-49c7-aa57-3a279c4dc10a.mp4

Has it something to do with, something you said earlier?
> Basically the library does some stuff in the background to keep the refresh rate up at the expense of perceived colour depth: lsbMsbTransitionBit

mrfaptastic, 2023-01-30:
Does that happen at 100% brightness?

I can't see what would cause this as there's no particular treatment of any row over any other row.

Could be an optical illusion?

Kouzeru, 2023-01-30:
Well, no, it's not optical illusion at all.
And yes, I'm testing this explicitly in 100% brightness.

Guess, let's see how is in yours?

It's the same HueSpectrumDemo except few parts in the loop changed
```ino
void loop() {
  // Canvas loop
  float t = (float)(millis()%4000)/4000.f;
  for(int x = 0; x < PANEL_RES_X; x++){
    float f = (abs(sin((float)((millis()%32000)/32000.f-(float)x/PANEL_RES_X/32.)*2.f*PI)))*255;
    float r =max(min(cosf(2.f*PI*(t+(float)x/PANEL_RES_X+0.f/3.f))+0.5f,1.f),0.f);
    float g =max(min(cosf(2.f*PI*(t+(float)x/PANEL_RES_X+1.f/3.f))+0.5f,1.f),0.f);
    float b =max(min(cosf(2.f*PI*(t+(float)x/PANEL_RES_X+2.f/3.f))+0.5f,1.f),0.f);
    for(int y = 0; y < PANEL_RES_Y; y++)
      if(y*2 < PANEL_RES_Y){
        float t = (2.f*y)/PANEL_RES_Y;
        dma_display->drawPixelRGB888(x,y,
          (r*t)*f,
          (g*t)*f,
          (b*t)*f
        );
      }else{
        float t = (2.f*(PANEL_RES_Y-y))/PANEL_RES_Y;
        dma_display->drawPixelRGB888(x,y,
          (r*t+1-t)*f,
          (g*t+1-t)*f,
          (b*t+1-t)*f
        );
      }
  }
}
```

Kouzeru, 2023-02-07:
nevermind, I just realized the problem, which is actually due to the loose connection on the row selectors.
E.g. having 8th rows strips, caused by the loose connection of the C-pin connection, hence it responses slower.

closed 2023-02-07 by Kouzeru