mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #387: ESP32 crashes when enabling a second serial port

mrfaptastic, 2023-02-01:
### Discussed in https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/discussions/368

<div type='discussions-op-text'>

<sup>Originally posted by **nospamsam2011** January  9, 2023</sup>
Good day All,

I was curious if anyone else has experienced startup issues when creating second serial port on the ESP32?  I ask because as soon as I create a second serial port (Serial1 or Serial2), the ESP32 crashes.  The IO pins do not conflict with the HUB75 pins and so I figured before I start digging deep, I would see if anyone else had a similar experience and if so what they had to do to enable a second serial port.

Thanks in advance!</div>

nospamsam2011, 2023-02-05:
The link you provided indicates the the discussion has been deleted... Can you provide a different link or why was the discussion deleted?

mrfaptastic, 2023-02-05:
Discussion has been moved to an issue

mrfaptastic, 2023-02-05:
Can you provide a sample sketch that causes this crash so I can replicate / test?

nospamsam2011, 2023-02-14:
Good day mrfaptastic,

My apologies for the response delay!  Digging deeper I found that the 2nd serial port issue is dependent upon the Rx/Tx pins selected.  If I use GPIO's 32/35 then the serial port crashes.  To rule out any issue with your library, I created a simple sketch which is simply the Serial and Serial2 ... and sure enough the GPIO Pins selected for Tx/Rx can crash the system.  Consequently, the Serial2 issue has nothing to do with your library.

My apologies for mistakenly correcting this issue!

Cheers,

Sam

nospamsam2011, 2023-02-14:
As a follow up to this and if anyone else get's caught by this issue...  Although the ESP32 names pins as GPIO, some pins are input only.  Effectively GPIO Pins from GPIO34 and up can only be used as inputs.  In my case I had GPIO35 set as serial2's Tx (output) which caused the system crash.  This is annoying, as these pins should have been called GPI insted of GPIO... Sadly, this means a board change to remap the Tx pin to another pin... argh...

You can see the various pin functionality on Page 60 of the ESP32 technical Reference Manual (Revision 4.8).

Cheers,

Sam

DarrylStrong, 2023-02-15:
Lots of useful information on this site ...
https://randomnerdtutorials.com/esp32-pinout-reference-gpios/
In particular that page for GPIO.
Cheers
Daz