mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #388: Display seems brighter on PlatformIO then on ESP-IDF

dylanmsu, 2023-02-02:
Hi,

When i was migrating my project from PlatformIO (Arduino framework) to ESP-IDF (IDF), i noticed a slightly lower brightness when i compiled with ESP-IDF. Both projects have exact same code and I set for both projects the 'setBrightness8' to 255. Tried the same with 'setPanelBrightness'. The brightness is still different. 

It may be noteworthy that setting the brightness on PlatformIO did not have any effect on the brightness whatsowever. That did work on ESP-IDF.
Also, the refresh frequency appears to be much lower on ESP-IDF than on PlatformIO

my setup:
- ESP32-pico-mini-02
- ESP-IDF 4.4

mrfaptastic, 2023-02-03:
Bizzare. Can you set the core debug log level to 'Debug' and then print the library debug output here for both implementations?

DarrylStrong, 2023-02-04:
If you have an oscilloscope then looking at the OE line may shed some light? (excuse the pun)

dylanmsu, 2023-02-05:
Here is the log output of both idf and platform io:
[esp32_log_platformio.txt](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/files/10610770/esp32_log_platformio.txt)
[esp32_log_idf.txt](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/files/10610769/esp32_log_idf.txt)

Unfortunately i don't have an oscilloscope available atm.

mrfaptastic, 2023-02-06:
idf:

```
I (432) I2S-DMA: Allocating 8192 bytes memory for DMA BCM framebuffer(s).
I (439) I2S-DMA: Minimum visual refresh rate (scan rate from panel top to bottom) requested: 60 Hz
W (449) I2S-DMA: lsbMsbTransitionBit of 0 gives 114 Hz refresh rate.
I (456) I2S-DMA: Recalculated number of DMA descriptors per row: 128
I (463) ESP32/S2: Allocating memory for 1024 DMA descriptors.
W (473) ESP32/S2: Creating final DMA descriptor and linking back to 0.
```

pio:
```
[   108][I][esp32_i2s_parallel_dma.cpp:438] allocate_dma_desc_memory(): [ESP32/S2] Allocating memory for 512 DMA descriptors.
[   118][D][esp32_i2s_parallel_dma.cpp:468] allocate_dma_desc_memory(): [ESP32/S2] Allocating 6144 bytes of memory for DMA descriptors.
[   132][W][esp32_i2s_parallel_dma.cpp:524] create_dma_desc_link(): [ESP32/S2] Creating final DMA descriptor and linking back to 0.
```

Seems to indicate they aren't outputting with the same colour depth etc. hence the difference. Hard to say what the root cause without looking at the sketch - perhaps you have configured different clock rates?

Also, please use the latest git version for both and see if that makes any difference. 

> It may be noteworthy that setting the brightness on PlatformIO did not have any effect on the brightness whatsowever. That did work on ESP-IDF.

That shouldn't be the case. so for example, setPanelBrightness(0) still shows output? I'd expect a black screen.

fanks8888, 2023-02-14:
Upload the migrated IDF full code to your GIT_ HUB, help you analyze it

dylanmsu, 2023-02-18:
Okay i think i found the culprit. the version was not actually the same. The project on IDF used the latest commit (at the time) of the master branch and PlatformOI used the latest release (3.0.2).

I tried upgrading the version on PlatformIO from 3.0.2 to 3.0.5 and the behaviour suddenly became the same as IDF. I guess there is a bug in this version which causes the color depth and the refresh frequency to be altered.

I also tried the latest commit (48d89f9d3f48aad7d9551425c7c48d9b9ab90671) and the colors are back to normal now.

mrfaptastic, 2023-02-19:
Thanks for verifying. 

Just to confirm though: You aresaying the current git version of this library resolves the issue? 

It's probably time to release a new version given all the excellent contributions that  @Kouzeru provided.

dylanmsu, 2023-02-19:
Yes indeed the issue seems to be fixed on the current master branch. I tested it again and took some pictures.

[led matrix at master](https://user-images.githubusercontent.com/16963581/219909294-887734ad-be3e-4187-866f-374af8443120.jpg)
[led matrix at v3 0 5](https://user-images.githubusercontent.com/16963581/219909290-41b2c64a-facd-45f8-9077-e866834d033e.jpg)
[led matrix at v3 0 2](https://user-images.githubusercontent.com/16963581/219909289-fefab33a-04b3-4c3e-bca8-482f9037c8b9.jpg)

fanks8888, 2023-02-19:
You are great. It is difficult to find problems like this. Can you share the optimized code with me? In addition, I want to make a library similar to HUB08 interface recently. Are you interested in discussing or giving suggestions together

closed 2023-02-19 by mrfaptastic

dylanmsu, 2023-02-19:
@fanks8888 

Here is a minimal project replicating the problem:
[RGB_LED_MATRIX_PIO_TEST.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/files/10776780/RGB_LED_MATRIX_PIO_TEST.zip)

You can change the version in platformio.ini. Just make sure that platformio actualy changed the version by checking '.pio\libdeps\esp32dev\ESP32 HUB75 LED MATRIX PANEL DMA Display\library.properties'. I noticed pio doesnt fetch the specified version sometimes.

The panel i'm using is a 32x32 with 4 simultaious scanlines (3bit line address) so i had to do a pixel remap in the code.

fanks8888, 2023-02-19:
@Dylan Missuwe
Dear Dylan Missuwe, thank you for your code, but I really want to see the ESP32-HUB75 code of the IDF version after your migration.

dylanmsu, 2023-02-19:
My apologies, i misunderstood.
Here is the migrated IDF project.
[RGB_LED_MATRIX_IDF_TEST.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/files/10777042/RGB_LED_MATRIX_IDF_TEST.zip)

After imprting the library, i had to outcomment the following lines to get it working:
- `src/platforms/${target}/gdma_lcd_parallel16.cpp` in file `CMakeLists.txt`
- `#include <Arduino.h>` in file `esp32_i2s_parallel_dma.cpp`

fanks8888, 2023-02-19:
Dear Dylan Missuwe, thank you very much. I received your ESP32_ HUB75_ IDF, I just compiled and tested with IDF, and there were some errors. I'm trying to find out the reason why I configured the environment myself. If there is a problem I can't solve myself, I will ask you again tomorrow