mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #392: Deeper color depth?

Kouzeru, 2023-02-07:
I'd like to have deeper color depth beyond 8-bit, so I can represent the panel richer shades of gamma-corrected colors.
The obvious black clipping on the 8-bit bother me much, and I'd like to reuse the 16-bits color correction table on the site referenced in the source, and modify the routines to make use it. (this can be enabled only when color correction is on)

board707, 2023-02-07:
I doubt that the color depth can be increased significantly...
Each additional bit will decrease the FPS about two times... and timings in these matrices is packed to the maximum already.

Kouzeru, 2023-02-07:
I disagree upon how it doubles the rendering period for each extending bit, at least as far I had been experimenting how the DMA's color data is structured. At least I did it anyway.

mrfaptastic, 2023-02-07:
Was it as simple as changing the COLOR_DEPTH define value to 9+ to get this additional colour depth?

I've never experimented with this.

mrfaptastic, 2023-02-07:
> I can submit the PR right away :)

I guess you've already done the work then? I don't give any guarantees that I'll merge such a PR if its a lot of change

Kouzeru, 2023-02-07:
I... don't know whether you like it or not, but let's see

board707, 2023-02-08:
> Because the reachability of depth bits heavily dependent on the horizontal resolution and the global panel brightness value `setBrightness`.

It seems to me that you talking about something different than colour depth.
What is connection between colour depth and horizontal resolution??
Do you know how colour depth is created in such matrices? Did you heard about the BCM model?

Kouzeru, 2023-02-08:
> What is connection between colour depth and horizontal resolution??

The period of how long OE for each row. E.g if the horizontal resolution is 64, we only have OE ranged from 0 to 63 each row.

Yes, I'm familiar with BCM. seems the driver do something like this, to my understanding, 
[image]
where true BCM performed for only the first several bits, determined by `lsbMsbTransitionBit`

Though after reading the source on the last, i think the driver performs this:
[image]
as stated
> Binary code modulation is somewhat like PWM, but easier to implement in our case. First, we define the time we would refresh the display without binary code modulation as the 'frame time'. For, say, a four-bit binary code modulation, the frame time is divided into 15 ticks of equal length.
>    We also define 4 subframes (0 to 3), defining which LEDs are on and which LEDs are off during that subframe. (Subframes are the same as a  normal frame in non-binary-coded-modulation mode, but are showed faster.)  From our (non-monochrome) input image, we take the (8-bit: bit 7 to bit 0) RGB pixel values. If the pixel values have bit 7 set, we turn the corresponding LED on in subframe 3. If they have bit 6 set, we turn on the corresponding LED in subframe 2, if bit 5 is set subframe 1, if bit 4 is set in subframe 0.
>    Now, in order to (on average within a frame) turn a LED on for the time specified in the pixel value in the input data, we need to weigh the subframes. We have 15 pixels: if we show subframe 3 for 8 of them, subframe 2 for 4 of them, subframe 1 for 2 of them and subframe 1 for 1 of them, this 'automatically' happens. (We also distribute the subframes evenly over the ticks, which reduces flicker.)

board707, 2023-02-08:
Thanks for your detailed answer.
I have already corrected my post :)
Yes, I understand what you mean, I myself use the principle of adding color depth by controlling brightness in my code for HUB75 panels.

But it still reduces FPS, although not twice. And if there are many such levels, the overall brightness of the matrix suffers.

Kouzeru, 2023-02-08:
I think doing this would preserve the most of the brightness (up to 95.3125%)
[image]

Just an idea, I don't know whether the driver does this or not. Most probably not, I think.

board707, 2023-02-08:
You should controlling brightness of least significant color bits, so your picture is quite incorrect. Your color planes 0-3 will be the longest.

Kouzeru, 2023-02-08:
I think you need to look at the picture again, and see the labels where it points to

Btw, to control the overall brightness:
[image]

board707, 2023-02-08:
Ok, I understood.
Firstly I don't mention that you distribute the levels between frames.

PR #393 (2023-02-08): Add support for color depth beyond 8-bits
resolves #392

This PR would enable deeper color depth support up beyond 8-bits, to express gamma corrected colors more deeply.

There is no benefit having more than 8-bits with `NO_CIE1931` enabled, as the raw colors always stay 8-bits. 

The default color depth is still 8-bits, hence everything should behave the same.

I've updated the example of the HueSpectrum, where it fades to black, back and forth, to see the effect of gradual shading, and also to easily compare between color depths.
closed 2023-02-08 by mrfaptastic