mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #401: Display rotated 180deg

tsctrl, 2023-02-19:
Hi @mrfaptastic , 

i just downloaded master and tried. Without any configuration change. the display was rotated 180 Degree from the previous 2.0.6 library version. I have change phase and use different driver but did not fix the issue. What was the update for this feature or where can i look for the fix?

Thanks,
TS

tsctrl, 2023-02-19:
update: 
only happen for single 1x1 panel configuration with virtual matrix panel. did not occur for chained panel. chained panel was correct as per existing configuration and behavior (serpentine, top down).

thanks!

mrfaptastic, 2023-02-19:
What happens with setRotate(true) or setRotate(false)?

tsctrl, 2023-02-20:
hi @mrfaptastic ,

single 1x1 panel configuration:
1. no setRotate in setup() after initialization cause the orientation to 180degree from the correct orientation (flip/upside down).
2. setRotate(true) in setup() after initialization cause the orientation to 90degree from the correct orientation(display is on right side and rotated vertically).
3. setRotate(false) in setup() after initialization is same as no 1.

2x3 panel configuration:
setRotate(true) in setup() after initialization cause the orientation to 270degree (display is on left side and rotated vertically).

correct orientation by definition is the correct display, on old version library where the input pin and the controller are at the right side of the display. i guess set rotate does not effected by this issue.

mrfaptastic, 2023-03-18:
Hi @drvkmr - Hopefully you can find the code again to help implement the rotation. setRotate(bool) was a temporary hack which we can remove.

drvkmr, 2023-03-22:
@mrfaptastic I have updated the rotation here, can you have a look and let me know what you think?
I can then do a pull request.
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/commit/53ecf54c56babf9d557ff76e0eef254755199060

mrfaptastic, 2023-03-22:
Hi @drvkmr - Thanks for this contribution. A couple of things.

Please rename the function to 'setRotation'  so it's aligned to Adafruit_GFX - I know this will break compatibility, but that is fine.

I need to test this with some physical panels, but I think we are missing the part where we recalculate the VirtualResX and Y, and swap the rows and columns around etc. as well... as  the code does the x/y co-ord mapping, what happens if somebody has a non-square chain of panels? The x or y co-ords would simply be out of bounds then and ignored? I don't know - perhaps you can test its use in the real word with say a 2x3 grid of panels or something.

https://github.com/adafruit/Adafruit-GFX-Library/blob/5ea4be3ffeae918899aeec8cbb67f87359cf0026/Adafruit_GFX.cpp#L1316

closed 2023-03-27 by mrfaptastic