mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #402: Large Flickering on Double Buffer

tsctrl, 2023-02-19:
Hi @mrfaptastic 

I have flickering issue with noticeable large duration of black being displayed before the lib able to feed the pixel back with the correct data. noticeable by eyes and more noticeable when rendering rgb bitmap.

```
if (flipBuffer) {
  dma_display->flipDMABuffer();
  dma_display->clearScreen();
}
```

seems like the display was swap before the pixel was properly plotted.
lib is using Adafruit library, did not happen on 2.0.6 library version.

tsctrl, 2023-02-19:
the reason i am updating library is due to panic issue in the library v2.0.6 as below:

```
======================== THREADS INFO =========================
  Id   Target Id          Frame
* 1    process 1073472652 0x40081a55 in panic_abort (details=0x3ffb0d5a \"abort() was called at PC 0x4013a918 on core 0\") at esp_system/panic.c:402
  2    process 1073483172 0x400d907a in flipDMABuffer (this=0x3ffc70e8) at esp32_led_matrix_dma.h:540
  3    process 1073478008 0x40194da0 in ?? () at ../components/app/src/app.cpp:77
  4    process 1073483532 0x401d3eae in trc_calc_duration () at freertos_tasks_c_additions.h:50
  5    process 1073507628 0x4012fdb6 in i2c_master_read (cmd_handle=0x3ffc6ac8, data=0x3ffc3334 <error: Cannot access memory at address 0x3ffc3334>, data_len=1, ack=I2C_MASTER_LAST_NACK) at driver/i2c.c:1238
```

the old mechanism to clear DMA buffer in v2.0.6 library version is having shared flag to check if the buffer was cleared. but the approach to do a while loop in the flip buffer to wait for the buffer to clear before returning true was causing system panic.

code that cause panic in v2.0.6:
```
// Double buffering flipping
// Flip to a buffer: 0 for bufa, 1 for bufb
// dmadesc_a and dmadesc_b point to the same memory if double buffering isn't enabled.
void i2s_parallel_flip_to_buffer(i2s_port_t port, int buffer_id) {

    if (i2s_state == NULL) {
      return; // :-()
    }

    lldesc_t *active_dma_chain;
    if (buffer_id == 0) {
        active_dma_chain=(lldesc_t*)&i2s_state->dmadesc_a[0];
    } else {
        active_dma_chain=(lldesc_t*)&i2s_state->dmadesc_b[0];
    }

    // setup linked list to refresh from new buffer (continuously) when the end of the current list has been reached
    i2s_state->dmadesc_a[i2s_state->desccount_a-1].qe.stqe_next = active_dma_chain;
    i2s_state->dmadesc_b[i2s_state->desccount_b-1].qe.stqe_next = active_dma_chain;

    // we're still shifting out the buffer, so it shouldn't be written to yet.
    //previousBufferFree = false;
	i2s_parallel_set_previous_buffer_not_free();
}

bool i2s_parallel_is_previous_buffer_free() {
    return previousBufferFree;
}
```

i was looking for the updated library where i could try the new approach on how the display was cleared. but looking the approach was to simply paint screen to black was causing the screen to be flickering

`
inline void clearScreen() { updateMatrixDMABuffer(0,0,0); };
`
this should be address properly @mrfaptastic, like the approach known before to use the showDMABuffer() and to have a method exposed by the library if ( !display->backbuffready() ) return; that can be use by user to properly timed the buffer clearing.

mrfaptastic, 2023-02-19:
The old flickering issue strikes again. There is never going to be a clean solution for this, as the DMA hardware is completely asynchronous to the Arduino sketch and at no stage are the timings guaranteed. 

But, I am confused – are you saying that it was working OK with 2.0.6 and prior, but now it is not? Just trying to understand where the regression started.

mrfaptastic, 2023-02-19:
The old flickering issue strikes again. There is never going to be a clean solution for this, as the DMA hardware is completely asynchronous to the Arduino sketch and at no stage are the timings guaranteed.... but this doesn't mean I am going to give up trying to find a solution!

I have pushed a few changes to the repository. Not sure if they fix the flickering, but it shouldn't cause a memory read fatal error.

Kouzeru, 2023-02-19:
As for me, I also do face the flicker.

Mind to put at the end of setup() `dma_display->flipBuffer()`?

tsctrl, 2023-02-20:
hi @mrfaptastic ,

the display still have flickering but lesser than before. (before can say around 3 flicker a sec) now is 1 every 10 sec.
but the display become very slow. it reduces the speed by half.

mrfaptastic, 2023-02-20:
What's the size of the display out of curiosity? 

So I'm still confused, in 2.0.6 you did or did not get flicker? And the problem with 2.0.6 was that there was a panic now and again?

mrfaptastic, 2023-02-20:
To confirm, with:
* 2.0.6 you didn't get flicker, but you got a panic?

I have made a change to 2.0.6 (2.0.7 legacy branch) which I think will fix for the panic condition you were having. Try going back to the 'old' version to see if it solves for your issue.

tsctrl, 2023-02-20:
ya, panic issue is solved. i just add yield there. wondering if i can update to 3.0.5 to have s3 capability. and yes 2.0.6 has no flickering only panic. flicker is 3.0.5 which i tested with result above

tsctrl, 2023-02-20:
from my observation, the flicker was due to high speed flip buffer. and look like the black timings are more longer than the suppose to be the proper pixel lit.

means the display was displaying black without the pixel draw to be ready in place.  more prominent with drawrgb bitmap function.

mrfaptastic, 2023-03-07:
I'm going to work on a whole new way of doing this secondary buffer thing.

This "flipping the DMA chain pointer" approach doesn't work because the DMA hardware does it's own thing and the software that runs on the CPU will never be able to synchronise perfectly.

primus192, 2023-03-07:
It was not happening to me on 3.0.3 though. On 3.0.5 i got a problem with
flipping the buffer working every second time i call it.

mrfaptastic, 2023-03-07:
What hardware? A standard ESP32?

primus192, 2023-03-07:
Yes, the standard one.

mrfaptastic, 2023-03-10:
Try the latest git master. @Lukaswnd made a commit to fix things.

primus192, 2023-03-10:
I can confirm that usage of doublebuffer is acting properly now. Everything is working for me as exprected, the buffer is not skipped :) Also no flickering for me.

primus192, 2023-03-10:
May i ask about orientation of the display? I am using six 64x64 panels (SM5166PF, they are working on FM6124 setting tho) connected as serpentine where second row is turned upside down like in readme. I attach pictures:

[image]
[image]

As you can see in this orientation everything is working fine. The ribbon which is connecting rows together is very long and connectors are far apart from each other. The only other way  is to connect them like so with short cable: 
[image]

but then the display will look like so (sorry im to lazy to reconnect them again and again)
[image]

There is my question. Can i somehow change the placement of xy origin or switch rows with one another? Or do you have any idea how can i connect it using short cable and be able to correctly display it?

mrfaptastic, 2023-03-10:
Are you using the virtualmatrix wrapper class? Per the documentation: https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/tree/master/examples/ChainedPanels

You should never need to use a long cable like that. The virtual matrix class should solve for the rotation of that 'row' of panels.

primus192, 2023-03-10:
For sure:

```
/*--------------------- RGB DISPLAY PINS -------------------------*/
#define R1_PIN 0 //25
#define G1_PIN 18 //33
#define B1_PIN 2
#define R2_PIN 14
#define G2_PIN 12
#define B2_PIN 13
#define A_PIN 23
#define B_PIN 19
#define C_PIN 5
#define D_PIN 17
#define E_PIN 32
#define LAT_PIN 4
#define OE_PIN 15
#define CLK_PIN 16
#define BUTTON 39
/*--------------------- MATRIX LILBRARY CONFIG -------------------------*/
#define SERPENT true
#define TOPDOWN false
#define NUM_ROWS 2
#define NUM_COLS 3
#define PANEL_RES_X 64     // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 64     // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN NUM_ROWS*NUM_COLS      // Total number of panels chained one to another
#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>
#include "Icons.h"
MatrixPanel_I2S_DMA* dma_display = nullptr;
VirtualMatrixPanel* virtualDisp = nullptr;

U8G2_FOR_ADAFRUIT_GFX* u8g2;
void setup() {
 if (device.config.mtx) {
        HUB75_I2S_CFG::i2s_pins _pins = { R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN };
        HUB75_I2S_CFG mxconfig(
            PANEL_RES_X,   // width
            PANEL_RES_Y,   // height
            PANEL_CHAIN,   // chain length
            _pins,   // pin mapping
            HUB75_I2S_CFG::FM6124      // driver chip
        );
        mxconfig.i2sspeed = mxconfig.HZ_20M;
        mxconfig.min_refresh_rate = 20;
        if (device.config.wifi) {
            mxconfig.double_buff = false;
        }
        else if (device.config.useGsm) {
            mxconfig.double_buff = true;
        }
        dma_display = new MatrixPanel_I2S_DMA(mxconfig);
        dma_display->begin();
        virtualDisp = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, SERPENT, TOPDOWN);
        measurement.begin();
        measurement.loop();
        u8g2 = new U8G2_FOR_ADAFRUIT_GFX();
        u8g2->begin(*virtualDisp);
        redrawMatrix();
    }
}

void redrawMatrix() {
    dma_display->setBrightness8(255);
    virtualDisp->clearScreen();
    u8g2->setFont(u8g2_font_6x12_me);
    u8g2->setForegroundColor(virtualDisp->color565(255, 255, 255));

    drawCentre("CIŚNIENIE", 32, 10, false, 0);

    //drawCentre("WILGOTNOŚĆ", 94, 10, false, 0);
    //drawCentre("TEMPERATURA", 160, 10, false, 0);

    //
    drawCentre("TEMP. WEWN.", 100, 25, false, 0);
    drawCentre("TEMP. ZEWN.", 100, 10, false, 0);
    //
    u8g2->setForegroundColor(virtualDisp->color565(0, 255, 0));

    String tempc;
    String tempextc;
    String humidp;
    String presshpa;

    tempc = (String)round1(measurement.getTemperature());
    tempc = tempc.substring(0, tempc.indexOf(".") + 1 + 1) + "°C";

    //
    tempextc = (String)round1(tempTable[0]);
    tempextc = tempextc.substring(0, tempextc.indexOf(".") + 1 + 1) + "°C";   
    //

    humidp = (String)round1(measurement.getHumidity());
    humidp = humidp.substring(0, humidp.indexOf(".") + 1 + 1) + "%";

    presshpa = (String)round1(measurement.getAltPressure(device.config.alt));
    presshpa = presshpa.substring(0, presshpa.indexOf(".")) + " hPa";

    u8g2->setFont(u8g2_font_7x13B_mf);
    
    //
    drawCentre(tempextc, 160, 10, false, 0);
    //

    drawCentre(tempc, 160, 25, false, 0);
    //drawCentre(humidp, 94, 25, false, 0);
    drawCentre(presshpa, 32, 25, false, 0);

    u8g2->setFont(u8g2_font_6x12_me);

    //LOGO
    //drawXbm565(64, 0, 64, 32, czpow[0], virtualDisp->color565(90, 140, 255)); // logo
    //LOGO

    virtualDisp->drawLine(0,28,192,28,virtualDisp->color565(255, 255, 255));
    virtualDisp->drawLine(0,75,192,75,virtualDisp->color565(255, 255, 255));

    u8g2->setForegroundColor(virtualDisp->color565(255, 255, 255));
    drawCentre("KWOTA WYPŁACONYCH DOTACJI", 96, 39, false, 0);
    u8g2->setFont(u8g2_font_7x13B_mf);
    u8g2->setForegroundColor(virtualDisp->color565(90, 140, 255));
    drawCentre((String)device.config.num1 + " PLN", 96, 50, false, 0);
    u8g2->setForegroundColor(virtualDisp->color565(255, 255, 255));
    u8g2->setFont(u8g2_font_6x12_me);
    drawCentre("LICZBA ZŁOŻONYCH WNIOSKÓW", 96, 61, false, 0);
    u8g2->setForegroundColor(virtualDisp->color565(90, 140, 255));
    u8g2->setFont(u8g2_font_7x13B_mf);
    drawCentre((String)device.config.num2, 96, 72, true, 0);
    u8g2->setForegroundColor(virtualDisp->color565(255, 255, 255));
    u8g2->setFont(u8g2_font_6x12_me);
    drawCentre("JAKOŚĆ POWIETRZA", 96, 88, false, 0);
    drawCentre("PM10", 32, 97, false, 0);
    drawCentre("PM2.5", 160, 97, false, 0);
    u8g2->setForegroundColor(colors[measurementTab._PM10tab[0]]);
    drawCentre("µg/m3", 32, 126, false, 0);
    u8g2->setFont(u8g2_font_7x13B_mf);
    drawCentre((String)value100Table[0], 32, 113, true, 0);
    u8g2->setForegroundColor(colors[measurementTab._PM25tab[0]]);
    drawCentre((String)value25Table[0], 160, 113, true, 0);
    u8g2->setFont(u8g2_font_6x12_me);
    drawCentre("µg/m3", 160, 126, false, 0);
    drawXbm565(75, 90, 40, 35, icon_bits[measurementTab._PMtab[0]], colors[measurementTab._PMtab[0]]);
    delay(100);
    virtualDisp->flipDMABuffer();
}

mrfaptastic, 2023-03-10:
Ok, so basically what it seems like is the 'botton up' serpentine chaining is broken.

Can you please run the basic ChainedPanels example (with the test pattern) based on the bottom-up chaining you want with the short cable, and send me a photo? I need to understand the exact panel rotation and then look at the code.

primus192, 2023-03-10:
Yes i will try and send you the results

primus192, 2023-03-10:
@mrfaptastic 

Long cable setup:

[image]
[image]

Short cable setup (code is the same) :

[image]

[image]

As you can see, lines do not match

mrfaptastic, 2023-03-10:
I take it for the long cable (top down chain) photo it's the following:

```
#define SERPENT true
#define TOPDOWN true
```
and shot cable photo

```
#define SERPENT true
#define TOPDOWN false
```
?

Can you run the short cable (bottom up chain)  example with the 'top down' config... I know it will produce a mess but will help visually:

```
#define SERPENT true
#define TOPDOWN true

```

primus192, 2023-03-10:
Every example i showed you was serpent true topdown false. With long cable the output is ok. Now i will show you short cable with true true:

[image]

primus192, 2023-03-10:
Connection, pinology, esp placement do not change when i rearrange those displays so i think it is still serpent true topdown false with long and short cable. So no fix in the software will get me to the right result. The construction of the displays cause that the connectors are far apart from each other..

mrfaptastic, 2023-03-10:
No No. This can be fixed in software easily.  I will need to find some old panels to test this.

primus192, 2023-03-10:
Thanks for bothering. To clarify i think both bottom up and top down are
working right. When i switch placement of esp and connect the rows together
on the other side it is working. Only that i need this long cable :) So i
think it would require to implement another function not to rotate but to
switch even rows.

mrfaptastic, 2023-03-10:
You're right. I need to have top down from left, top down from right, bottom up getting left, bottom up from right.

Never thought the starting left/right location of the ESP32 device would matter... But they're you go.

Just to confirm, in the last photo with the short cable the esp is behind the panel on the bottom right ?

primus192, 2023-03-10:
@mrfaptastic 
BOTTOM:
[image]

Im flipping to the led side like this:
[image]

So it becomes:
SERPENTINE TRUE TOPDOWN FALSE:
[image]

SERPENTINE TRUE TOPDOWN TRUE:
[image]

mrfaptastic, 2023-03-12:
@primus192 - I ended up going down a rabbits hole and re-engineering the VirtualMatrixPanel class.

It will now support this chaining using the CHAIN_TOP_LEFT_DOWN.

Refer to the updated 'ChainPanels' example and documentation. These changes break backward compatability. If you have issues, please raise a new issue specific to this issue. Thanks.

closed 2023-03-12 by mrfaptastic

primus192, 2023-03-12:
Thank you for it. I will test that on monday :)

mrfaptastic, 2023-03-12:
Thank you. Please let me know if it works as well.

mrfaptastic, 2023-03-12:
Also, where do you get those plastic clips to connect the panels together? Have you custom made those with a 3d printer? Can I buy some off you?

primus192, 2023-03-12:
> Also, where do you get those plastic clips to connect the panels together? Have you custom made those with a 3d printer? Can I buy some off you?

Yes, the joints are designed and 3d printed .

I currently have three of them for 64x64 p3 panels like so:

Joint for middle fixing four 64x64 panels:
[image]
Joint for fixing edges of columns:
[image]
Joint for fixing edges of rows:
[image]

Do you have 3d printer so i can send you the files or do you want me to send some to you (im from Poland)? Please mind that different density and size of the panels have different distances between screws and it depends also how you chain them. Please let me know the details and the distances of your threaded inserts that you would need. I also can create "pill shaped" screw holes so you will be able to fit it as you want.

mrfaptastic, 2023-03-12:
> Do you have 3d printer so i can send you the files or do you want me to send some to you (im from Poland)? Please mind that different density and size of the panels have different distances between [...]

Wow that's amazing. I don't have a 3D printer, but I can probably find somebody local (in the UK) if you have a 3d model that's of use? 

I have a bunch of P3 panels that I use to regression test this library with. 46mm and 22mm seem to be the important lengths.

[image]

primus192, 2023-03-13:
> Wow that's amazing. I don't have a 3D printer, but I can probably find somebody local (in the UK) if you have a 3d model that's of use?
>
> I have a bunch of P3 panels that I use to regression test [...]

@mrfaptastic I tried the newest version of the library. It is working in all orientations now :) It is nice that now you can even rotate the display by changing CHAIN_TOP_LEFT_DOWN to CHAIN_BOTTOM_RIGHT_UP because they are somehow interchangable :) It works with long and short cable configuration so all configurations possible. 

Meanwhile, thank you very much again and i am attaching the joints files that i have prepared for your 46x22 spaces.
[46x22fix.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/files/10955756/46x22fix.zip)

tsctrl, 2023-03-17:
@mrfaptastic ,

on my end i still could see a blink once every 1-2 sec, did not have issue on gif and jpg rendering. however the changes are workable and less than before, probably i need to retest by removing delay at some other place and try again, but i doubt it will fix this issue.

https://user-images.githubusercontent.com/75555993/225950131-e1894b43-e35b-49f6-8ff3-f83a9dcabc9f.mp4

primus192, 2023-07-05:
> > > Do you have 3d printer so i can send you the files or do you want me to send some to you (im from Poland)? Please mind that different density and size of the panels have different distances between screws and it depends also how you chain them. Please let me know the details and the distances of your threaded inserts that you would need. I also can create "pill shaped" screw holes so you will be able to fit it as you want.
> > 
> > 
> > Wow that's amazing. I don't have a 3D printer, but I can probably find somebody local (in the UK) if you have a 3d model that's of use?
> > I have a bunch of P3 panels that I use to regression test this library with. 46mm and 22mm seem to be the important lengths.
> 
> @mrfaptastic I tried the newest version of the library. It is working in all orientations now :) It is nice that now you can even rotate the display by changing CHAIN_TOP_LEFT_DOWN to CHAIN_BOTTOM_RIGHT_UP because they are somehow interchangable :) It works with long and short cable configuration so all configurations possible. I find it very stable with double buffer finally working as expected.
> 
> Meanwhile, thank you very much again and i am attaching the joints files that i have prepared for your 46x22 spaces. [46x22fix.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/files/10955756/46x22fix.zip)

As i do not want to open new issue i want to ask here. I am using outdoor displays now and have connected them the same way as indoor panels above. I use four scan example as it is working for outdoor panels:

```
#define NUM_ROWS 1
#define NUM_COLS 1

#define PANEL_RES_X 64     // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 64     // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN NUM_ROWS*NUM_COLS      // Total number of panels chained one to another
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_NONE

void setup() {
        HUB75_I2S_CFG::i2s_pins _pins = { R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN };
        HUB75_I2S_CFG mxconfig(
            PANEL_RES_X*2,              // DO NOT CHANGE THIS
            PANEL_RES_Y/2,   // height
            PANEL_CHAIN,   // chain length
            _pins,   // pin mapping
            HUB75_I2S_CFG::FM6124    // driver chip
        );
        mxconfig.i2sspeed = mxconfig.HZ_20M;
        mxconfig.min_refresh_rate = 10;
        mxconfig.double_buff = true;
        mxconfig.clkphase = false;
        dma_display = new MatrixPanel_I2S_DMA(mxconfig);
        dma_display->begin();
        virtualDisp = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, VIRTUAL_MATRIX_CHAIN_TYPE);
        virtualDisp->setPhysicalPanelScanRate(FOUR_SCAN_64PX_HIGH);
        measurement.begin();
        measurement.loop();
        u8g2 = new U8G2_FOR_ADAFRUIT_GFX();
        u8g2->begin(*virtualDisp);
        redrawMatrix();
    }
```
The panels are working but arrangement of the rows is wrong and i need to use long ass cable again with them. Is the new type of virual matrix chain type arrangement ommited in code when using FOUR_SCAN_64PX_HIGH or am i missing something?