mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #405: cannot see gifs with code

fragmation, 2023-03-01:
I tried this code and cannot see any gif :( any help please ?

// library includes
#define FILESYSTEM SPIFFS
#include <SPIFFS.h>
#include <AnimatedGIF.h>
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <Fonts/FreeSans9pt7b.h>
#include <Fonts/FreeSansBold12pt7b.h>

MatrixPanel_I2S_DMA *dma_display = nullptr;
// ----------------------------
#include "SD.h"
#include "SPI.h"

//===============Pins Configuration=====================
//======================================================
// my pin config
// Micro SD Card Module Pinout
// VCC = 5V
// GND = GND

#define HSPI_MISO 21
#define HSPI_MOSI 19
#define HSPI_SCLK 18
#define HSPI_CS 2

// Led Panel Pinout
#define R1_PIN 25
#define G1_PIN 26
#define B1_PIN 27
#define R2_PIN 14
#define G2_PIN 12
#define B2_PIN 13
#define A_PIN 23
#define B_PIN 22
#define C_PIN 5
#define D_PIN 17
#define E_PIN 32
#define LAT_PIN 4
#define OE_PIN 15
#define CLK_PIN 16

// Configure for your panel(s) as appropriate!

#define PANEL_RES_X 64 // Number of pixels width of each individual panel module.
#define PANEL_RES_Y 64 // Number of pixels height of each individual panel module.

#define NUM_ROWS 1 // Number of Vertical rows of chained panels
#define NUM_COLS 1 // Number of Horizontal rows of chained panels

#define PANEL_CHAIN NUM_ROWS*NUM_COLS // Total number of panels chained one to another

// Change this to your needs, for details on VirtualPanel pls see ChainedPanels example
#define SERPENT false
#define TOPDOWN false

SPIClass *spi = NULL;
AnimatedGIF gif;
File f;
int x_offset, y_offset;

// Draw a line of image directly on the LED Matrix
void GIFDraw(GIFDRAW *pDraw) {
uint8_t *s;
uint16_t *d, *usPalette, usTemp[320];
int x, y, iWidth;

iWidth = pDraw->iWidth;
if (iWidth > MATRIX_WIDTH)
[... 250 lines cut ...]
char fname[256];
File root, temp;

while (1) {// run forever
//root = FILESYSTEM.open(szDir); // File System
root = SD.open(szDir); // For SD card
if (root) {
temp = root.openNextFile();
while (temp) {
if (!temp.isDirectory()) {// play it
strcpy(fname, temp.name());

      Serial.printf("Playing %s\n", temp.name());
      Serial.flush();
      ShowGIF((char *)temp.name());
    }
    temp.close();
    temp = root.openNextFile();
  }
  root.close();
} // root
delay(100); // pause before restarting
} // while
}
using a sd card, a P3 64x64 panel and gifs resized to 64x64 pixels any clue ?

mrfaptastic, 2023-03-10:
Refer to: https://github.com/bitbank2/AnimatedGIF/blob/master/examples/ESP32_LEDMatrix_I2S/ESP32_LEDMatrix_I2S.ino

fragmation, 2023-03-10:
thanks mr faptastic, with that work but nned more space to put the gifs :(  and in those example dont use the sd card

mrfaptastic, 2023-03-14:
Refer to updated examples in examples folder now.

closed 2023-03-18 by mrfaptastic