mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #406: Need to fix the file HardwareSerial.cpp

MichaelAndronov, 2023-03-02:
Need to fix the file HardwareSerial.cpp
as in this video
[https://www.youtube.com/watch?v=GwShqW39jlE&t=203s](url)
[image]

_Originally posted by @MichaelAndronov in https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/issues/387#issuecomment-1450244887_

Lukaswnd, 2023-03-09:
Do you guys know, that you can set the pins for the HardwareSerial?
The function is part of the HardwareSerial class.
`void setPins(int8_t rxPin, int8_t txPin, int8_t ctsPin = -1, int8_t rtsPin = -1);

Serial1.setPins(4, 21);
Serial1.begin(115200);`

You could also set them in Serial.begin(). Just make sure so set the config as well.
`void begin(unsigned long baud, uint32_t config=SERIAL_8N1, int8_t rxPin=-1, int8_t txPin=-1, bool invert=false, unsigned long timeout_ms = 20000UL, uint8_t rxfifo_full_thrhd = 112);

Serial1.begin(115200, SERIAL_8N1, 4, 21);`

I don't know why these are the default pins. They are connected to the flash (and SPIRAM/PSRAM if present), which makes it really hard to actually use them especially with uart.

MichaelAndronov, 2023-03-10:
OK!
Great, thanks a lot for the advice. I will try.
Michael