mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #414: Serpent-less arrangment

drvkmr, 2023-03-14:
Just saw the new updates with the non-standard chaining and ma loving it. So much more clear and precise.
The only issue I realised instantly on updating is that it's not possible to chain panels in a serpent-less configuration. Am happy to work on implementing this change if you can help out @mrfaptastic, unless this is already possible and I am just missing it.

mrfaptastic, 2023-03-14:
Thanks for the positive feedback! Glad that you love it!

Yeah I had burnt enough of my weekend on the update and took the assumption that nobody would use non-serpent chaining as it would require some long cables... But clearly I am wrong. 🤣

I'll take a look at it tonight as I think I can implement this quickly with another parameter to the constructor to serpentine.

mrfaptastic, 2023-03-14:
Where are you going to put your ESP device on the back of the grid of panels (that is, where will the start point me)? Can it be the top right?

I can easily build support for a 'top right zigzag down' config. Couldn't be bothered doing the other three permutations.

drvkmr, 2023-03-15:
It's bottom right in this setup, but if you do a partial implementation I can try and complete it.

mrfaptastic, 2023-03-15:
Try with the esp32-dev branch of this library. Have added CHAIN_TOP_RIGHT_DOWN_ZZ

mrfaptastic, 2023-03-15:
Sorted. Refer to Dev branch for the details

drvkmr, 2023-03-15:
Cheers, I'll have a go this week.

commit fe5d4148 (2023-03-15): Add ZigZag options. Close #414

CHAIN_BOTTOM_RIGHT_UP_ZZ
CHAIN_TOP_RIGHT_DOWN_ZZ
closed 2023-03-16 by mrfaptastic