mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #415: ICN2037BP will it work?

iceblader2013, 2023-03-15:
Hey I got a P4 panel as you can see in the pictures but the labeling is different and I can't find the drivers in the list under running. Can you measure the pin assignment by where D and E is?

[image]
[image]
[image]
[image]

DarrylStrong, 2023-03-15:
The icn2037 are basic shift register drivers so should be fine.

Regarding d and e, test with a meter to see if the standard pins are connected to ground. I suspect that the legend is incorrect.

Cheers

Darryl

iceblader2013, 2023-03-15:
what do you mean by standard pins on ground? ground on the plug or on the power connection?

iceblader2013, 2023-03-15:
all pins are connected to ground

DarrylStrong, 2023-03-15:
What re the other drive ics on the board?

There are the hc245, icn and what others?

iceblader2013, 2023-03-15:
[image]

iceblader2013, 2023-03-15:
the other all the same

DarrylStrong, 2023-03-15:
There must be drivers for the rows, are there ics under the plastic moulding?

iceblader2013, 2023-03-16:
hey under the plastic is 8 ics ICN2012

DarrylStrong, 2023-03-16:
Good, those are 3 to 8 line demultiplexers.
[http://wiki.amperka.ru/_media/%D0%BF%D1%80%D0%BE%D0%B4%D1%83%D0%BA%D1%82%D1%8B:rgb-led-matrix-64x32:icn2012_datasheet.pdf]url

Therefore for a 32:1 scan there must be 2 other row selects D and E.

This is the pinout most likely used.
[https://external-preview.redd.it/qA2JTWsbdDTG-HRStneUEd-kLK65XU8AfLjVzjyO82w.jpg?auto=webp&s=332c9721fceefa13a6073e2b744336236f33b621](url)

board707, 2023-03-16:
> Have you a ideas ?

Swap the R G B pins?

iceblader2013, 2023-03-16:
which are the rgb pins? ^^

board707, 2023-03-16:
ESP32 pins, connected to R1 G1 B1 and R2 G2 B2 pins of HUB75 connector

mrfaptastic, 2023-03-18:
Change the physical wiring or simply swap the GPIO/PIN/COLOR pin assignment in your code because your physical panel has been incorrectly made.

closed 2023-03-18 by mrfaptastic