mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #416: Four_Scan_patten not working

KvarTechnologies, 2023-03-16:
Hi ,
Hardware:
P10 RGB (32x16 pixel)
DOIT ESP32 DEV kit1

I am trying to program Four scan pattern , its giving me following errors 

Arduino: 1.8.19 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None, Disabled"

In file included from C:\Users\admin\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src/ESP32-HUB75-MatrixPanel-I2S-DMA.h:12,
                 from C:\Users\admin\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\examples\Four_Scan_Panel\Four_Scan_Panel.ino:18:
C:\Users\admin\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src/platforms/platform_detect.hpp:58:19: note: #pragma message: Compiling for original ESP32 (released 2016)
   #pragma message "Compiling for original ESP32 (released 2016)"
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\admin\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\examples\Four_Scan_Panel\Four_Scan_Panel.ino: In function 'void setup()':
Four_Scan_Panel:94:122: error: no matching function for call to 'VirtualMatrixPanel::VirtualMatrixPanel(MatrixPanel_I2S_DMA&, int, int, int, int, bool, bool)'
     FourScanPanel = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, SERPENT, TOPDOWN);
                                                                                                                          ^
In file included from C:\Users\admin\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\examples\Four_Scan_Panel\Four_Scan_Panel.ino:23:
C:\Users\admin\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src/ESP32-VirtualMatrixPanel-I2S-DMA.h:74:3: note: candidate: 'VirtualMatrixPanel::VirtualMatrixPanel(MatrixPanel_I2S_DMA&, int, int, int, int, PANEL_CHAIN_TYPE)'
   VirtualMatrixPanel(MatrixPanel_I2S_DMA &disp, int _vmodule_rows, int _vmodule_cols, int _panelResX, int _panelResY, PANEL_CHAIN_TYPE _panel_chain_type = CHAIN_TOP_RIGHT_DOWN)
   ^~~~~~~~~~~~~~~~~~
C:\Users\admin\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src/ESP32-VirtualMatrixPanel-I2S-DMA.h:74:3: note:   candidate expects 6 arguments, 7 provided
C:\Users\admin\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src/ESP32-VirtualMatrixPanel-I2S-DMA.h:66:7: note: candidate: 'constexpr VirtualMatrixPanel::VirtualMatrixPanel(const VirtualMatrixPanel&)'
 class VirtualMatrixPanel : public Adafruit_GFX
       ^~~~~~~~~~~~~~~~~~
C:\Users\admin\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src/ESP32-VirtualMatrixPanel-I2S-DMA.h:66:7: note:   candidate expects 1 argument, 7 provided
C:\Users\admin\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src/ESP32-VirtualMatrixPanel-I2S-DMA.h:66:7: note: candidate: 'constexpr VirtualMatrixPanel::VirtualMatrixPanel(VirtualMatrixPanel&&)'
C:\Users\admin\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master\src/ESP32-VirtualMatrixPanel-I2S-DMA.h:66:7: note:   candidate expects 1 argument, 7 provided
Using library ESP32-HUB75-MatrixPanel-DMA-master at version 3.0.6 in folder: C:\Users\admin\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-DMA-master 
Using library Adafruit-GFX-Library-master at version 1.11.3 in folder: C:\Users\admin\Documents\Arduino\libraries\Adafruit-GFX-Library-master 
Using library Adafruit_BusIO-master at version 1.11.3 in folder: C:\Users\admin\Documents\Arduino\libraries\Adafruit_BusIO-master 
Using library Wire at version 2.0.0 in folder: C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\libraries\Wire 
Using library SPI at version 2.0.0 in folder: C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\libraries\SPI 
exit status 1
no matching function for call to 'VirtualMatrixPanel::VirtualMatrixPanel(MatrixPanel_I2S_DMA&, int, int, int, int, bool, bool)'

Pls Do let me know how do i resolve this

tsctrl, 2023-03-17:
line 94: 
`FourScanPanel = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, SERPENT, TOPDOWN);`
change to:
`FourScanPanel = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y);`

KvarTechnologies, 2023-03-18:
thanks for the reply the display now gets compiled and get program too but it shows green scrolling part as shown in the video bellow. Pls do guide me to work it out.
I am using p10 RGB slab (32x16 [pixcel)](
https://user-images.githubusercontent.com/104092776/226098192-a6da9b28-9d72-468c-bd1f-bec313486c47.mp4
)

board707, 2023-03-18:
> p10 RGB slab (32x16 pixcel)

Your info is not complete. The next important parameter of panel is a scan. 
Please show the image of reverse side of the panel with model ID and other text clear readable.

KvarTechnologies, 2023-03-18:
P10 Slab (32x16 )
Chip: SM16208SC
[image]
[image]

mrfaptastic, 2023-03-18:
Please use the latest git master version of this library, the example and the VirtualMatrixPanel class has been updated to fix a related issue.

closed 2023-03-18 by mrfaptastic

KvarTechnologies, 2023-03-18:
i have updated the library, and ran the sample code of Four scan. The output is as follows 

https://user-images.githubusercontent.com/104092776/226104619-3ca25934-8a1a-44d6-9b2b-c480dfa688f2.mp4

KvarTechnologies, 2023-03-18:
I have update the provided library. The output still remains the same. 

https://user-images.githubusercontent.com/104092776/226104794-729c00d6-fbb8-4854-a2fe-df8676bb16a6.mp4

DarrylStrong, 2023-03-18:
Looks like the driver chips are combined row and column and therefore not compatible with the library.

Datasheet here.

https://datasheetspdf.com/datasheet/SM1620B.html