mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #417: 1 panel 128*64

Maestros95, 2023-03-16:
Hello my Friend

Congrats for your work.

i would like to ask you how can i connect 1 panel 128*64 with esp32. I have connected the panel with wires such here, but how can i connect the pin E?

#define R1 25
#define G1 26
#define BL1 27
#define R2 14
#define G2 12
#define BL2 13
#define CH_A 23
#define CH_B 19
#define CH_C 5
#define CH_D 17
#define CH_E -1 // assign to any available pin if using two panels or 64x64 panels with 1/32 scan
#define CLK 16
#define LAT 4
#define OE 15

Are correct the defines?

#define PANEL_RES_X 128 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 64 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW
#define PANEL_CHAIN NUM_ROWS*NUM_COLS    // total number of panels chained one to another

// Change this to your needs, for details on VirtualPanel pls see ChainedPanels example
#define SERPENT false
#define TOPDOWN false

Thanks!!!

board707, 2023-03-16:
> #define CH_E -1 

You should connect a real pin to E channel for your panel work

Maestros95, 2023-03-16:
OK.

I connected on pin 33.

#define CH_E 33

But all define pins lines are comments.

board707, 2023-03-16:
> But all define pins lines are comments

Why did you considered it as comments? By "#" mark?

Maestros95, 2023-03-16:
[image]

board707, 2023-03-16:
Try 4_HueValueSpectrumDemo.ino example - it contains a full set of pins available for setup

Maestros95, 2023-03-16:
Ok.
I set up the code about my panel.

Can i show pictures jpeg, bitmap or png photo on the screen on my panel?

Maestros95, 2023-03-16:
In project ChainedPanelsAuroraDemo how can i set the pins? such as 4_HueValueSpectrumDemo.ino example.

Maestros95, 2023-03-16:
> be

In ChainedPanelsAuroraDemo how can I set the pins?

closed 2023-03-18 by mrfaptastic