mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #425: 160 X 80  resolution led panel

Jiwno, 2023-03-19:
I'm trying to use 160 X 80 pixel led panel but it doesn't work at all. Arduino code uploaded without any error but the output video of led is just white color with some strange grain. Is there anyone who have the same issue of mine?

board707, 2023-03-19:
Please clarify - the size of panel is 160x80 mm or 160x80 pixels? Could you please show the photo of reverse size of it with ID and and other text clearly readable?

Jiwno, 2023-03-19:
https://ko.aliexpress.com/item/1005004752444756.html?spm=a2g0o.order_list.order_list_main.5.3382140fJlKGbG&gatewayAdapt=glo2kor
[image]

This is the of the product and it's 160x80 pixels.

board707, 2023-03-19:
According to description it seems to me that the panel has s-PWM driver, incompatible with the library. Could you write models of the chips?

Jiwno, 2023-03-19:
I'm using esp-wroom-32 chip for Arduino.

board707, 2023-03-19:
No, please indicate the models of chips in the matrix

Jiwno, 2023-03-19:
oh, sorry. It says sm16380sc azbqysaa51 on the chip

mrfaptastic, 2023-03-19:
Same issue as #416 it seems. Not supported chip as its a totally different way of operation.

Jiwno, 2023-03-19:
thank you for helping me